sparkfun/SparkFun_RTK_Firmware #91: Horizontal Accuracy .vs. Mean 3D Standard Deviation

LeeLeahy2, 2022-04-03:
### Subject of the issue
I have not found a good description between the horizontal accuracy and mean 3D standard deviation in the documentation.  On the RTK Express display and in the system menu the horizontal accuracy is being displayed.  In the base menu, the mean 3D standard deviation is set to 5.000 meters.  With a weak signal (indoors, not near a window), the horizontal accuracy displays around 2 meters (2.2 - 1.55), but the serial output displays: Waiting for Horz Accuracy < 1.00 meters: 1.84

I have not found a place where I can display the mean 3D standard deviation.  If the mean 3D standard deviation is the metric being used for the transition from no-fix to fix then it would be useful if that value was displayed in addition to the horizontal accuracy.  If horizontal accuracy is also a criteria, then would it make sense to allow value be adjustable in the base menu?

System menu displays: SIV: 31, HPA (m): 1.841, Lat: 21.356980710, Lon: -158.081943974, Altitude (m): 229.0

### Your workbench
* What version of RTK Surveyor firmware are you running?  RTK_Surveyor_Firmware_v1_11-RCApr1.bin
* Are you connected to the device over Bluetooth?  No, using USB connected to Config ESP32 port
* What app?  minicomm  (terminal emulator)
* Are you transmitting NTRIP back to the device?  No, NTRIP server configured, but base station not transmitting yet because sufficient accuracy not met
* How is everything being powered?  Battery / USB
* Are there any additional details that may help us help you?

### Steps to reproduce
1. Place receiver indoors where reception is not great.
2. Turn on the RTK Express
3. Configure the base station to use Survey-In mode
4. Use the setup button to enter base mode

### Expected behavior
1. Transition from no-fix to fix after meeting criteria in base menu.
2. Adjust all criteria for no-fix to fix transition in base menu.
3. Display actual values for all criteria while waiting for no-fix to fix transition

### Actual behavior
1. Mean 3D standard deviation value not displayed, unclear if this criteria for no-fix to fix transition has been met.
2. Horizontal accuracy criteria not settable in base menu, serial output indicates that this is fixed at 1.00 meter.
3. Horizontal accuracy criteria is not met due to intentionally weak signal
4. Not able to verify NTRIP transmission since fix is not possible

nseidle, 2022-04-04:
The 'wait for 1m HPA' is a requirement that we put in place after lots of base testing. If the ZED is put into base mode immediately upon power up it can often lead to 3D Means that never drop below the required 5m. I believe it's because the mean gets feed with some *very* large inaccuracies when the ZED is first getting a lock. To prevent this, we wait for a reasonable position lock (HPA < 1m) before starting a base survey in. 

Said differently: The small amount of time to get a 3D fix (<30s) + the 60s of default base survey in time was significantly less than the 5 to 10 minutes we were seeing if the base survey in process was started from warm or hot start.

This has the intended side-effect of preventing a survey-in process during poor reception, ie, indoors. You'd never get a proper survey indoors so the survey never starts.

We've had a previous request (#76) to allow modification of the 1m HPA requirement. You can modify the 'surveyInStartingAccuracy' but I don't recommend changing it from the default of 1m.

closed 2022-04-04 by nseidle

gdt, 2022-04-04:
From an outsider (who doesn't have one of your devices yet), brief comments about why I'm perplexed reading this.  (I have an F9P and have been doing RTK with MassDOT's network, and have never tried to survey in.)

  - There are lots of ways to give horizontal accuracy (standard deviation, CEP, 95%, 99%).  really standard deviation isn't accuracy, and I know the receiver is estimating accuracy per fix, which is different from the actual accuracy.
  - I find "mean standard deviation" very confusing, because while it seems likely the mean is over the accepted samples (those with HPA < 1m), it's not immediately obvious what data is being put into a standard deviation calculation.  It doesn't really make sense to talk about a standard deviation of one sample.   I certainly understand the point of wanting a bunch of consistent samples.   And, surely the samples over a minute aren't independent; tropo delay error is surely very correlated.
  - It's extra confusing to use horizontal as a start gate and then 3D as an acceptance criteria.
  - I don't know if this is with SBAS or without, and how that affects if the results are WGS84(G2139) vs ITRF2014 (or whatever WAAS is in; it's really hard to find documentation).  For that matter, I don't know how the F9P combines 4 constellations that are in 4 different datums.  Probably these days (vs in the 90s) none of this matters.   But SBAS if it has tropo corrections might get better ellipsoidal heights.   I know people tend to not want SBAS turned on for RTK, but surveying in for RTK is perhaps different.
 
gpsd has gpsprof, and what it does is tricky.  There is CEP50/CEP95/CEP99, but it's not based on the true position, but the average of the samples - which is really something different.   And then there's a MIL standard which is about finding similar values, but by scaling the standard deviation assuming the distribution is normal.    My attempt to clarify the code's comments a bit is at https://github.com/ukyg9e5r6k7gubiekd6/gpsd/commit/3a96fcc558934f5ef3755e8b6d4fa0a65abdd5a9 in case that is helpful.

nseidle, 2022-04-04:
@LeeLeahy2 I added a submenu under Base to configure the initial positional accuracy. Documentation in the HGs will come after a firmware release.

[image]

@gdt - Thanks! It's always good to hear from experts in the field. 

> I find "mean standard deviation" very confusing

We use 'Mean 3D Standard Deviation' in the RTK firmware, and it comes from u-blox's documentation and nomenclature.

[image]

For now, we use the initial fix accuracy requirement to reduce overall Survey-In time and reduce the chance of Survey-In failure (we've seen multiple instances where the module gets stuck if mean 3D is too large at start). If, some day, you do a Survey-In, please feel free to recommend improvements.