sparkfun/SparkFun_RTK_Firmware #93: New min_spiffs.csv causes software reset loop

LeeLeahy2, 2022-04-05:
### Subject of the issue
Step 3 of the updated firmware directions (Change the Parition table) causes a software reset loop following build and upload of RCApr4 to the RTK Express.

### Your workbench
* What version of RTK Surveyor firmware are you running?   Building and running RCApr4
* Are you connected to the device over Bluetooth?   No, connected via USB to the Config ESP32 port
* What app?   minicomm (terminal emulator)
* Are you transmitting NTRIP back to the device?   No
* How is everything being powered?  Battery / USB
* Are there any additional details that may help us help you?  No

### Steps to reproduce
1. Follow the firmware installation and build directions from https://github.com/sparkfun/SparkFun_RTK_Firmware/tree/release_candidate/Firmware
2. Replace the min_spiffs.csv file as indicated in step 3 of the directions.
3. Build and upload the image to the RTK Express

### Expected behavior
The firmware should boot successfully

### Actual behavior
The firmware enters a software reset loop!  See
[20220404_1641_RCApr4.txt](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/8414524/20220404_1641_RCApr4.txt)

nseidle, 2022-04-05:
Did you set the 'Flash Size' to 16MB (128mbit)? Step 5 got added late last night.

LeeLeahy2, 2022-04-05:
That was the missing step.