sparkfun/SparkFun_RTK_Firmware #94: Bluetooth LE for GPS data?

XPav, 2022-04-05:
I am looking at integrating some devices with the RTK Facet via Bluetooth.  Obviously, RFCOMM over Classic Bluetooth is there today, and with some client-side implementation, I can happily parse NMEA messages.

With the industry pushing to Bluetooth Low Energy though (because it's "better"), I'm wondering if it makes sense to add this to the Facet?

I see u-blox has their own "Low Energy Serial Port Service" https://content.u-blox.com/sites/default/files/u-connectXpress-LowEnergySerialPortService_ProtocolSpec_UBX-16011192.pdf, and I see the test sketch for "OpenLog" in the repo -- are there any other possible standards that could be used?

nseidle, 2022-04-05:
It's very much been on my mind as this is the backdoor into Apple iPhones (they don't support SPP without magic ICs that require licensing fees). 

The ESP32 is quite capable of BLE. I just haven't had a good target app to design to. If anyone has a recommendation for a good surveying app that requires BLE and has all the UUIDs made public I will certainly make a go of it.

I *think* SW Maps also supports BLE so I may ping @avinabmalla.

avinabmalla, 2022-04-10:
@nseidle SW Maps supports BLE using the Nordic serial profile. Having an option to switch between SPP and BLE would be great for users developing devices based on BLE that interfaces with the SparkFun RTK receivers. 

You can use my library to implement BLE.

[https://github.com/avinabmalla/ESP32_BleSerial](https://github.com/avinabmalla/ESP32_BleSerial)

It also works with iOS! You can use the AdaFruit BlueFruit LE Connect app to connect and view the NMEA stream on an iPhone.

burnsed, 2022-05-11:
+1 
I believe this would also open up the device for use with browser based apps in Android via the [bluetooth API](https://web.dev/bluetooth/)

burnsed, 2022-07-10:
FYI, I have a [preliminary version written](https://github.com/sparkfun/SparkFun_RTK_Firmware/compare/main...infinite-tree:SparkFun_RTK_Firmware:main). I'm hoping to do some real world testing with it this week and hopefully submit a PR.

nseidle, 2022-07-10:
This looks very good @burnsed! Please PR when you're ready.

nseidle, 2022-08-18:
Huge shout out to @burnsed for adding BLE. Thanks!

closed 2022-08-18 by nseidle