sparkfun/SparkFun_RTK_Firmware #95: RTK Express shutdown quickly at power on

Eric-FR, 2022-04-08:
After firmware upgrade to v1.12 (from RTK_Surveyor_Firmware_v1_11-RCApr1), at power on, the splash screen with new version is visible  for about 1s and, then, display is switched off (and no serial connexion). 

With or without µSD card.

No way to use is now.

nseidle, 2022-04-08:
Hi Eric, Do not fear. Units are always recoverable.

> upgrade to v1.12

We don't have a v1.12. Did you mean [RTK_Surveyor_Firmware_v1_11-RCApr5.bin](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/download/v1.11/RTK_Surveyor_Firmware_v1_11-RCApr5.bin)?

You are using an RTK Express, correct? I will try to replicate. For now, if you want, you should be able to change the firmware back to v1.10 using the batch program instructions [here](https://github.com/sparkfun/SparkFun_RTK_Firmware/tree/main/Binaries). You will likely have to 'time it just right' where the batch file is ready to run, then you power on the unit and right when a COM port appears press enter to cause the batch file to run.

Sorry for the hassle! Thank you for testing.

Eric-FR, 2022-04-08:
Lol. Made an upgrade to RTK_Surveyor_Firmware_v12.bin...

I will try to go back to a reasonable firmware.

nseidle, 2022-04-08:
OH. Hah! Yes, that would cause a problem. 

Maybe I'll store some of those older binaries in a sub directory so that's harder to do.

Eric-FR, 2022-04-08:
I'm back to a correct firmware!!! Still, some warning when downgrading would be welcome.

nseidle, 2022-04-08:
> some warning when downgrading would be welcome.

Agreed. I'll think on this and try to find a good way.

closed 2022-04-08 by nseidle

gdt, 2022-04-08:
I didn't realize the firmware upgrade was all in the bootloader.   In that case, why wouldn't it just take putting in a card with the right firmware to unbrick?  But if you think it's impossible, I withdraw my suggestion since you obviously understand this ecosystem vastly better than me.

Eric-FR, 2022-04-08:
I can't upgrade from the SD card because the device was switching off too fast. With the command line, I was able to start the device and, then, start upgrade within a fraction of second.

nseidle, 2022-04-08:
@gdt - Sorry, I got snippy. But ya, it's a good idea, but I don't have a good way to implement.

@Eric-FR and my future self: There *is* a way to force firmware update actually. Rename a firmware file to *RTK_Surveyor_Firmware_Force.bin* and the unit will load it at power on, and then delete the file.