sparkfun/SparkFun_RTK_Firmware #96: Add NMEA over TCP

gdt, 2022-04-10:
There are various ways to have GNSS receivers provide data to computers, and it's good to meet interfaces that others might expect, as long as it doesn't get too hard or too complicated.

One method is just NMEA over a TCP socket, either a listening socket on the IP address of the receiver, or having the receiver connect to a configured address.   This is supported by Vespucci, an openstreetmap editor Android app, and by Ardusimple's NTRIP ESP32 module.   I believe it is also supported by RTKLIB, which is how vespucci grew the feature.

This would ideally involve three things:
  1) configuration to have the ESP32 listen on a configured port #, and when there's a TCP connection, just send the NMEA.   Obviously be graceful about errors.  Bonus points for multiple sockets at once.  This gets most of the value.
  1) configuration to have the ESP32 connect to an IP address and port and send NMEA, and retry on failure
  1) as previous but instead of an IP address, connect to the router, so that when your phone is the hotspot to let NTRIP fetch RTCM3, it will also be the place with the GIS program.  Could just be a toggle for router or configured addr.
  1) Maybe maybe, some support for dynamic DNS, but that's half baked so I am mentioning it but not really suggesting it.

Vespucci doesn't have Bluetooth/BLE support at the moment. It probably should, but the TCP socket request would be useful for lots of things, like having an RTK Explorer Express join a wifi and have some other node read/store the NMEA for long-term obs, perhaps as  part of base with QC, etc.

nseidle, 2022-09-06:
TCP over NMEA has been added thanks to @LeeLeahy2 in the latest RC. See basic instructions for using a cell phone AP + Vespucci [here](https://github.com/sparkfun/SparkFun_RTK_Firmware/pull/309). Pulling down RTCM over WiFi, then pushing TCP over WiFi is working well. While I'm not sure why someone would want to do it, you can also, simultaneously connect over BT and pull to SW Maps. Please give it a try and let us know if anything breaks.

@gdt - As the pro Vespucci user in the group, do you feel like documenting a 'how to' under the [GIS software](https://docs.sparkfun.com/SparkFun_RTK_Firmware/gis_software/) section?

Nightglobe, 2022-09-07:
The instructions are too misunderstood. We prefer an IP form TCP connection: Port
You connect the NTRIP client to the wifi from facet to select to transmit nmea  as a server to a port, predefined as in Vespucci or another port, there is more work with it ,  Bluetouth is ok in this form but in the field an extra consumer is spoiled. And if we still use WiFi it is preferable to be able to use TCP IP without having to use Bluetouth.

gdt, 2022-09-07:
Thanks to all that helped - it's great to see this.    I have a bunch of ticket responses to work through and agreed that I should add a howto to that section.    I have been able to get QField to work, too, at least briefly, and will add that too.

nseidle, 2022-09-07:
@Nightglobe - 

> WiFi it is preferable to be able to use TCP IP without having to use Bluetouth.

You can easily use WiFi for TCP NMEA broadcast, and WiFi for RTCM download. BT was just an extra thing I was testing. It's not needed.

Nightglobe, 2022-09-07:
It does not understand me wrong, it is good to be both connections and wifi and Bluetouth for topographic engineers depending on age or knowledge or technology, depending on the programs used, although I have passed 50, I use both, of both . For  Example I have colleagues and customers using Emlid RS2, with Ublox Z-F9P identical to those used in the wide range that a Sparkfun factory, it has a setting from the Reachwiu 3 program that allows GPS configuration as imput corrections on the client, or Bluetouth  ,or  radio, etc, and  as a transmission of positions on NMEA I use TCP/IP, Bluetouth. The WiFi and Bluetout chip is about the one used by Sparkfun and I do not see why the future may not be too  able SparkFun  product to be as good technically with them. I personally use a software produced by Geoelctron, Surpad or Surv X (another name of it) very similar to Survey the master I signaled in and proposed to use in a post on the forum (gabriel_r), https: // forum. sparkfun.com/viewtopic.php?f=116&t=57966

nseidle, 2022-09-21:
This has been implemented in [v2.5](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.5).

closed 2022-09-21 by nseidle

pyrog, 2022-10-25:
Tested successfully with [TcpGPS - Topografía con GNSS](https://play.google.com/store/apps/details?id=com.aplitop.tcpgps)

[image] [image]  [image]

Nightglobe, 2022-10-25:
CAN YOU EXPLAIN TO ME THE CONNECTION STEPS. I AM USING A PORTABLE WIFI ROUTER WITH A PHONE CARD THAT ASSIGNS AN IP TO THE GNSS DEVICE AND ALTHOUGH I KNOW WHAT IP IS ASSIGNED TO THE GNSS DEVICE I CANNOT CONNECT TO THAT PORT

Nightglobe, 2022-10-25:
FROM THE GPS SETTINGS IN THE TERMINAL, I ENABLED THE OTHER SETTINGS FOR NTRIP CLIENT, SOCKET ETC.

pyrog, 2022-10-25:
2 steps:
- connect the RTK device in WiFi
- connect the device to the software in TCP to send NMEA stream.

Which software are you using on your smartphone/ tablet?

Nightglobe, 2022-10-25:
SMARTFONE

Nightglobe, 2022-10-25:
SORY SURPAD ANG TCPGPS

Nightglobe, 2022-10-25:
MORE PRECISELY WHICH SPARKFUN GNSS RECEIVER ARE YOU USING, IT MAY BE A PROBLEM WITH MY RECEIVER IF IT FAILS, SPARKFUN  FACET GPS-19984

Nightglobe, 2022-10-25:
WHEN I CONNECT VIA WIFI TO THE NTRIP PROVIDER USING GPS-19984, IT GIVES ME THE NMEA CORRECTIONS RECEIVED VIA BLUETOOTH, SO IT IS ASSIGNED AN IP ADDRESS FROM THE ROOTER, FROM THE TERMINAL SETTINGS or wifi (192.168.4.1) I SET THE NTRIP CLIENT , FROM THE TERMINAL WITH THE KEY s , I GIVE ENABLE , c and s then exit , I try to access by connecting the controller to the same wifi network to the ip dedicated to the GNSS receiver, PORT 1958 AND NOTHING

pyrog, 2022-10-25:
@Nightglobe  could you please read rule 4 of [Netiquette](https://doepud.co.uk/netiquette) ? 😃

pyrog, 2022-11-03:
@carhargps and @Nightglobe could you share your settings please ?
I suspect that editing settings on the terminal don't save all parameters.

carhargps, 2022-11-04:
I'm using RTK Facet, firmware version v2.5 and Tera Term VT

[image]

[image]

I usualy enable debug message **s)-->d)-->31)** *Print NMEA TCP status* and outputs that is trying to connect.

pyrog, 2022-11-06:
I think that settings from the terminal is bugged. That is why I ask you to share your settings 😉 
(Just replace sensitives parameters by "secret")

#### Sparkfun RTK settings

[SFE_Surveyor_Settings_0.txt](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/9945803/SFE_Surveyor_Settings_0.txt)

```
ntripClient_CasterHost=caster.centipede.fr
ntripClient_CasterPort=2101
ntripClient_CasterUser=secret
ntripClient_CasterUserPW=secret
ntripClient_MountPoint=NCY

ntripClient_wifiSSID=secret
ntripClient_wifiPW=secret

enableNmeaClient=0
enableNmeaServer=1
```

#### [TcpGPS](https://play.google.com/store/apps/details?id=com.aplitop.tcpgps&hl=en&gl=US) settings

`192.168.43.204` is the local address of the Sparkfun RTK device.
`1958` is the port of the NMEA TCP server on the device.

[image]

Nightglobe, 2022-11-06:
I made the settings and they worked on TcpGPS
They didn't work on Surpad, but not everything can be perfect, it can still be improved, thanks for the settings you provided