sparkfun/SparkFun_RTK_Firmware #99: Documentation defects

Eric-FR, 2022-04-16:
On the page:
https://learn.sparkfun.com/tutorials/sparkfun-rtk-express-hookup-guide/firmware-updates-and-customization

```
Updating Firmware From the SD Card

From time to time SparkFun will release new firmware for the RTK Express to add and improve functionality. For most users, firmware can be upgraded by loading the appropriate firmware file from the [binaries repo folder](https://github.com/sparkfun/SparkFun_RTK_Firmware/tree/main/Binaries/For_SD_Loading) onto the SD card and bringing up the firmware menu as shown below:

```

The binaries repo folder link is incorrect (error 404).

Also note that last ublox F9P firmware version is 1.30. I saw somewhere in documentation that 1.13 was still indicated as the last one (can't find where now...).

Eric-FR, 2022-04-16:
For ublox firmware, for instance: https://learn.sparkfun.com/tutorials/setting-up-a-rover-base-rtk-system/all#updating-the-receiver-firmware

nseidle, 2022-04-16:
> The binaries repo folder link is incorrect 

Fixed! Thanks

> For ublox firmware, for instance:

Fixed. 

FYI - I am documenting the new v1.11 firmware features first to the Facet hookup guide. I'll get the changes duplicated over to Express/Surveyor shortly.

closed 2022-04-16 by nseidle