sparkfun/SparkFun_RTK_Firmware #100: Latest SF FW and uBlox FW Breaks Bluetooth

cgorski, 2022-04-19:
### Subject of the issue
Upgrading to 
RTK Express Firmware: v1.11-Apr 12 2022
ZED-F9P Firmware: HPG 1.30
Breaks bluetooth
v1.10 works fine

### Your workbench
Have an Express and Express+ - v1.11 on Express+ w latest ublox firmware also does not work

### Steps to reproduce
Tell us how to reproduce this issue. Please include a copy of your settings.txt file used on the unit.

### Expected behavior
Use SWMaps on Android or cat /dev/tty.bluetooth - should have data

### Actual behavior
No data

cgorski, 2022-04-19:
Additional info - Bluetooth connects both on sw maps kindle device as well as Mac book, but can't get data on either

nseidle, 2022-04-19:
Thanks for reporting! I am able to get it to fail as well, but not all the time. Seems to be worse after a full system reset. I will get digging and get you a binary shortly.

nseidle, 2022-04-19:
The error arose because after a factory reset, the ZED's UART gets reset to 38400bps. The ESP32 is expecting 460800 but v1.11 firmware was not correctly re-configuring the ZED. I've added the systems check to correctly detect and configure the ZED's UART. 

Additionally, I was a bit concerned that the changing of the [serialTimeout(0) for ESF](https://github.com/sparkfun/SparkFun_RTK_Firmware/commit/03d158f34ed9f41008f7cb4c820372f78c715993) could have caused some problems. I no longer think it is part of the problem, but I did move the GNSS serial timeout setting to a user reachable debug menu (9) and increased the default from 0ms to 1ms, just in case we need to modify it in the future. All is working well for me. Please give [v1.12 RC](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.12) a try and let me know if it fixes the issue.

cgorski, 2022-04-20:
Awesome! I’m taking the units out for some surveying work tomorrow but once I finish up I’ll give the RC a go.

nseidle, 2022-05-11:
I believe this has been fixed. Please re-open if you continue to see problems.

closed 2022-05-11 by nseidle