sparkfun/SparkFun_RTK_Firmware #101: Additional profiles lost after firmware upgrade

Eric-FR, 2022-04-22:
While using RTK_Surveyor_Firmware_v1_11-RCApr5, I defined four named profiles. They were available in the display through Settings menu.

After firmware update to RTK_Surveyor_Firmware_v1_11, profiles aren't visible. Connecting with serial and going to Profiles menu, I only see the first name. When changing to the second menu, name is only Default but, after leaving the Profiles menu, which induces a reboot, corresponding profile name (and parameters) is back. So, after parsing all profiles, I get them all back incl. names and parameters, and I can select them from menu settings in the display.

nseidle, 2022-05-05:
This is an unfortunate side affect of firmware updates. We remove all previous settings in order to avoid wrong or corrupt settings from being set. 

If this is a very large annoyance, please let me know and there is perhaps something we can do. It's not clean, but I may be able to change to a clear text, local, settings file. There's still a risk that a bad setting could sneak it but it's much lower.

Eric-FR, 2022-05-06:
No, it is not a large annoyance. The surprising thing is that I can indeed recover profiles by selecting them.

nseidle, 2022-05-11:
I do love these issues :) I start pulling on a thread and my sweater comes apart.

I believe I've successfully re-written the NVM functions to allow proper handling of profiles loaded from LittleFS (aka NVM) and the SD card (if detected).

It works like this (for my future self):

* Load the profile number (currently 0 to 3)
* Load the settings from LittleFS
* Load settings from SD if present and Platform_Settings_[ProfileNumber].txt is available
* Load the profile names into a global array (used for displaying the user's Setup menu on the OLED)

If there is a difference between the internal settings (LittleFS) and the SD card, the settings on the SD card overrule.

I'll get a RC out shortly.

nseidle, 2022-05-12:
@Eric-FR - Please give the [RCMay11](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.13) (or newer) release a try. Profiles, and profile differences between SD and internal memory, should now play a lot better together. Additionally, we no longer clear profiles when firmware is updated.

nseidle, 2022-05-17:
@Eric-FR - I've got a new RC for you - v2.0 over [here](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.0). Please give it a try.

closed 2022-05-17 by nseidle