sparkfun/SparkFun_RTK_Firmware #109: NTRIP Client documentation needed

LeeLeahy2, 2022-05-05:
### Subject of the issue
Missing/out of date documentation

### Your workbench
* What version of RTK Surveyor firmware are you running? v1.11
* Are you connected to the device over Bluetooth? No
* How is everything being powered? Battery and USB cable to Config ESP32 port
* Are there any additional details that may help us help you?  The Configure GNSS Receiver documentation in the https://learn.sparkfun.com/tutorials/sparkfun-rtk-express-hookup-guide is out of date.  It show Toggle SBAS as entry #4 and no following entries.  Entry #4 is currently Set Constellations and entry #5 is Toggle NTRIP Client.  Entries 6-14 show up after the NTRIP client is enabled.

### Steps to reproduce
1. Use a USB cable to connect from a PC to the Config ESP32 port.
2. Using a terminal emulator on the PC (Eg. on linux, minicom) connect to the USB serial port (115200, 8-bits, no-parity)
3. Enter a carriage return (Enter key)
4. After the main menu comes up, enter the number one (1)

### Expected behavior
The expected behavior is correct

### Actual behavior
The actual behavior is correct, but the menu displayed does not match the documentation.  There are also a number of undocumented items in the list, and more when the NTRIP client is enabled.

LeeLeahy2, 2022-05-06:
If I understand step 1 above, it should also state: On a PC or Mac, edit the settings file...  If this is the case, how do I initially get the settings file onto the SD card for editing?

Would you please also document how to do this using the serial configuration?  Using the serial debugger does not require the removal of the SD card from the RTK device.

pyrog, 2022-05-06:
1. Set the device
* From a computer by editing the config file
* From a terminal connected by USB (or BT ?)
* From the web browser

(Work in progress: will be edited later)

nseidle, 2022-05-17:
I've updated the Facet documentation. Once that looks good, I'll transfer it to the Express and Surveyor hookup guides.

Please review `Configure GNSS Receiver` under https://learn.sparkfun.com/tutorials/sparkfun-rtk-facet-hookup-guide#system-configuration---serial-terminal

gdt, 2022-05-19:
Documentation about using NTRIP over WiFi in rover mode is also missing.   in addition, the relationship between WiFi NTRIP client and BT is unclear and from reading the github notification stream not very well, it is surprising.

nseidle, 2022-06-01:
[NTRIP Client](https://sparkfun.github.io/SparkFun_RTK_Firmware/configure_gnss/#ntrip-client) and [NTRIP Server](https://sparkfun.github.io/SparkFun_RTK_Firmware/configure_base/#ntrip-server) modes are now documented in the new docs

closed 2022-06-01 by nseidle