sparkfun/SparkFun_RTK_Firmware #110: UBX TIME IN SD CARD

IngridBerru, 2022-05-06:
Hi,
I'm testing the RTK Express + and I saw that the UBX file created in the SD card has a creation time different from UBC or local time, for example, the local time is 17:00 Hrs, but when I see the last modified time of the UBX file it says that is 4:00 Hrs (It's no my local time neither the UTC nor something near those times), so I was wondering how I can configure this time so It matches the local time or the UBC time, so when I have different tests in one day I can tell to which time it corresponds.
Thank you in advance.

nseidle, 2022-05-11:
I assume UBC is your local time? If I'm wrong please correct me.

All logs have their 'date file access' tags updated with UTC time, every [5 seconds](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/81089c1602e609be82783012e0a471e243b27227/Firmware/RTK_Surveyor/RTK_Surveyor.ino#L552). 

An offset from UTC is doable. It would have to be in minutes and hours as there are time zones not a full hour from UTC. Is that a feature that you'd like? Does anyone else see the need for logs set to local time? I personally would prefer to use UTC just so I knew exactly when a log was made.

pyrog, 2022-05-12:
> I personally would prefer to use UTC just so I knew exactly when a log was made.

👍

nseidle, 2022-05-17:
Closing due to inactivity.

closed 2022-05-17 by nseidle