sparkfun/SparkFun_RTK_Firmware #111: Unable to use BlueTooth connection when NTRIP client is enabled

pyrog, 2022-05-06:
If the NTRIP client is enabled in the setting file, it is not possible to use BT.

### Your workbench
* RTK Surveyor firmware **v1.12-Apr 19 2022**
* Device connected over Bluetooth.

### Steps to reproduce
* Enable NTRIP client (`enableNtripClient=1`) in the default profile `SFE_Surveyor_Settings_0.txt`.
* Power on the device.
* On the phone, activate the BlueTooth and start the NTRIP client.

### Expected behavior
Even is the NTRIP client is enabled, the BT should be enabled immediately (preferred) or after a timeout if the device can't access the WIFI AP.

### Actual behavior
Bluetooth isn't enabled so the NTRIP client on the phone can't be connected to the device.

### Note
If the SD card with several profiles is not mounted on the device, only the default profile is loaded in flash.
So it's not possible to select another profile with i.e. the NTRIP client disabled, another mountpoint or caster…

On the field It's not easy to start the device AP, edit the settings with a web browser, leave the AP and connect it in BT.

nseidle, 2022-05-11:
Ah! I see. You don't want simultaneous BT + WiFi, you want the device to fail gracefully back to BT if WiFi fails... A good idea! I'll see what I can do.

nseidle, 2022-05-11:
> If the SD card with several profiles is not mounted on the device, only the default profile is loaded in flash. So it's not possible to select another profile with i.e. the NTRIP client disabled, another mountpoint or caster…

If no SD card is in the device, the device will use the profiles loaded into memory. You should be able to select between the profiles using the setup button.

gdt, 2022-05-11:
What I want is simultaneously:
  - connect to WiFi AP and do NTRIP
  - provide NMEA on bluetooth

This is *I think* what it will take to make this device usable with QField, which does not have NTRIP support.  Or with any other program that works with BT GPS by simply reading the NMEA stream.  I'm having trouble following this ticket -- if what I describe doesn't work and is out of scope for this ticket I can open a new one.

nseidle, 2022-05-12:
@gdt - That's good feedback. I'm still, say 60%, certain that *simultaneous* BT and WiFi will be a blocker, but I need to do some more testing.

nseidle, 2022-05-12:
@pyrog - Please give the [RCMay11](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.13) (or newer) release a try. NTRIP Client now fails gracefully and drops back to BT if the given SSID is not found.

pyrog, 2022-05-12:
Seem work fine, thanks 😃 

Full test outdoor tomorrow.

nseidle, 2022-05-17:
Closing because I believe it's working. Please re-open if you see any failures.

As a side note - I was able to successfully run the NTRIP Client and Bluetooth at the same time but if you do anything extra, the system will crash due to limited RAM. I think with some very careful RAM cultivation we *could* have a firmware in the future that simultaneously does both WiFi and BT, but I'm still not convinced it's a useful feature to have.

closed 2022-05-17 by nseidle

pyrog, 2022-06-28:
> but I'm still not convinced it's a useful feature to have.
Me too 😄 

Will see in next release how WIFI is useful and probably enough.

commit 7f6d596c (2022-05-12): Fix #111 - BT if no WiFi NTRIP Client