sparkfun/SparkFun_RTK_Firmware #118: Can't update firmware over WIFI

pyrog, 2022-05-12:
From [RTK_Surveyor_Firmware_v1_12.bin](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/download/v1.12/RTK_Surveyor_Firmware_v1_12.bin) to [RTK_Surveyor_Firmware_v1_13RC-May11.bin](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/download/v1.13/RTK_Surveyor_Firmware_v1_13RC-May11.bin).

Tested on RTK Surveyor and RTK Express.
Express display _"OTA could not begin"_

pyrog, 2022-05-12:
[HTML](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/main/Firmware/RTK_Surveyor/AP-Config/index.html), [CSS](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/main/Firmware/RTK_Surveyor/AP-Config/src/style.css) and [javascript](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/main/Firmware/RTK_Surveyor/AP-Config/src/main.js) code should be minified ?

nseidle, 2022-05-12:
> code should be minified ?

Yes in general. Months ago I left them clear text because it was changing so frequently but it has calmed down quite a bit now. I should really gzip them like the other files. That would save us about 86k. Certainly helpful.

The underlying issue is that all units up to this point have been programmed with partitions limited to 4MB even though there is a 16MB module inside. The >2MB binary can't fit into the 2MB OTA1 partition that is available. And you can't change the partitions via OTA. The fix is to load code over USB using the 'batch_program.bat' inside the [Binaries](https://github.com/sparkfun/SparkFun_RTK_Firmware/tree/main/Binaries) folder. This will force the partition changes.

We could cut and trim and squeeze the firmware to keep it under 2MB, but the real solution is to have users update via esptool.  You only have to do it once, then OTA is properly set. I will double check 

Hmm... I think it's time to break things and release a v2 firmware to show folks that there's a big change. I will repurpose the [Artemis Firmware Upload](https://github.com/sparkfun/Artemis-Firmware-Upload-GUI) tool to make it easy on users (and avoid scary things like command lines).

For now, I recommend using the instructions on the Binaries page to upload over serial.

nseidle, 2022-05-12:
I've pushed v2.0 of the firmware. Please see [v2.0-RCMay12](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.0).

pyrog, 2022-05-12:
Updated via WIFI both RTK Surveyor and Express.

Thank you very much Nathan  😄

closed 2022-05-12 by pyrog