sparkfun/SparkFun_RTK_Firmware #119: NMEA checksum error!

pyrog, 2022-05-12:
### Subject of the issue
Several NMEA checksum errors detected in log files.
Detected by JOSM or PaulZC [UBX/NMEA Integrity Checker](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/main/Utils/UBX_Integrity_Checker.py).

### Your workbench
* [RTK_Surveyor_Firmware_v1_12.bin](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/download/v1.12/RTK_Surveyor_Firmware_v1_12.bin)
* [SFE_Surveyor_Settings_0.txt](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/8681127/SFE_Surveyor_Settings_0.txt)
* [SFE_Surveyor_220506_112532.ubx](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/8681124/SFE_Surveyor_220506_112532.ubx.txt)

Usage: `python3 UBX_Integrity_Checker.py SFE_Surveyor_220506_112532.ubx `

nseidle, 2022-08-08:
Hi pyrog - Just wanted to let you know we're aware of this and are trying to get it fixed.

tonycanike, 2022-08-09:
Thank you Nathan! This should be a huge help.  

The PaulZC UBX/NMEA Integrity Checker finds numerous issues in the UBX files created with RTK Firmware version 2.3.

One 5 hour file had 26 successful resyncs and over 100 (I stopped counting at 100) NMEA checksum, UBX checksum, and NMEA excessive length errors reported by PaulZC's checker.

I've been having issues with RTKCONV (demo5_b34f1) when I convert the UBX files to RINEX.  It will skip over hours of data from the middle of a UBX file, resulting in large gaps in the OBS files I send to OPUS.  Which of course get rejected.

tonycanike, 2022-08-16:
Still getting lots and lots of errors when I check the UBX file, both NMEA checksums and UBX errors.  Here's the summary of the pages and pages of errors from UBX_Integrity_Checker.py

```
Processed 173345458 bytes
File size was 173345458
Longest valid UBX message was 2072 bytes
Longest valid NMEA message was 83 characters
Message types and totals were:
Message type: GNTXT   Total: 3
Message type: GNRMC   Total: 62423
Message type: GNGGA   Total: 62425
Message type: GNGSA   Total: 312131
Message type: GPGSV   Total: 95144
Message type: GLGSV   Total: 87933
Message type: GAGSV   Total: 78481
Message type: GBGSV   Total: 79800
Message type: GQGSV   Total: 15607
Message type: GNGST   Total: 62424
Message type: 0x02 0x15   Total: 62398
Message type: 0x02 0x13   Total: 276740
Number of successful resyncs: 50
```

Running RTK_Surveyor_Firmware_RC-Aug_15_2022.bin.

nseidle, 2022-08-19:
I think we're getting closer. This is a perfect test over 6 hours using RC Aug18. I was doing basic 5 sentence NMEA logging at 4Hz so nothing outlandish, just to make sure we have the basics. I'll increase to 

[image]

[image]
Above is the byte comparison to the ZED-F9P output. It takes a bit of log alignment, but once aligned, we are logging *exactly* what the ZED is outputting for 26 million bytes.

tonycanike, 2022-08-19:
That's excellent!  Thanks for the update.  

Been testing here too.   I downloaded the Aug 18 RC earlier today and collected for two hours, PPP Logging defaults, though the measurement rate was only every 5 seconds (0.2Hz) as trying to debug my OPUS processing, I wanted to reduce chances of logging corruption, and OPUS decimates to 30 sec intervals anyway.   As an aside, I just have too many trees here is the conclusion I'm reaching.

**ZERO** errors from the UBX_integrity_checker.py  for that run!

I'll start a run at 4Hz &  PPP logging defaults and let you know how that goes.

tonycanike, 2022-08-21:
Things are looking good!

My second test ran for over 5 hours the other day on my Facet. 

No errors reported by UBX_Integrity_Checker.p  !

4Hz measurement rate & PPP defaults (7 messages).
The UBX file was 202,962 KB (windows explorer size)

Here're the firmware versions from the output.
$GNTXT,01,01,03,v2.4-Aug 18 2022
$GNTXT,01,01,04,HPG 1.32

As an aside, after converting with RTKCONV and waiting 36 hrs +/-,  OPUS rejected it as too noisy.  Trees=bad.  TEQC and RTKPLOT showed noise and slips.   CSRS was quite happy to give me a solution though.  This is a separate issue I'm sorting out.

```
C:\....>python UBX_Integrity_Checker.py SFE_Facet_220819_202418.ubx
UBX Integrity Checker

Could this file contain any NMEA messages? (Y/n):
Do you want to repair the file? (y/N):

Processing SFE_Facet_220819_202418.ubx

NMEA $ found at byte 0. Attempting to process the message

Processed 207832232 bytes
File size was 207832232
Longest valid UBX message was 2008 bytes
Longest valid NMEA message was 83 characters
Message types and totals were:
Message type: GNTXT   Total: 3
Message type: 0x02 0x13   Total: 309155
Message type: 0x02 0x15   Total: 81062
Message type: GNRMC   Total: 81062
Message type: GNGGA   Total: 81062
Message type: GNGSA   Total: 405310
Message type: GNGST   Total: 81062
Message type: GPGSV   Total: 125394
Message type: GLGSV   Total: 97246
Message type: GAGSV   Total: 104462
Message type: GBGSV   Total: 93443
Message type: GQGSV   Total: 20265

Bye!
```
```

Stefal, 2022-08-25:
Hi !
With the August 18 RC, I still have a lot of checksum errors (RTK Surveyor).
My settings are :
- 5Hz
- default NMEA messages
- UBX RAWX
- UBX SFRBX

```
Panic!! UBX checksum error!
Sync lost at byte 3029007. Attemting to re-sync.
Panic! Maximum rewind attempts reached! Aborting...

Processed 3029008 bytes
File size was 36259802
FILE SIZE MISMATCH!!
Longest valid UBX message was 2104 bytes
Longest valid NMEA message was 85 characters
Message types and totals were:
Message type: GNTXT   Total: 3
Message type: 0x02 0x15   Total: 605
Message type: 0x01 0x07   Total: 628
Message type: 0x01 0x02   Total: 633
Message type: 0x01 0x14   Total: 636
Message type: GNRMC   Total: 638
Message type: GNGGA   Total: 638
Message type: GNGSA   Total: 3176
Message type: GNGST   Total: 632
Message type: 0x01 0x35   Total: 625
Message type: 0x01 0x43   Total: 602
Message type: 0x01 0x20   Total: 630
Message type: 0x01 0x21   Total: 632
Message type: 0x02 0x13   Total: 2145
Message type: GPGSV   Total: 864
Message type: GLGSV   Total: 783
Message type: GAGSV   Total: 637
Message type: GBGSV   Total: 505
Message type: GQGSV   Total: 126
Number of successful resyncs: 100
```

nseidle, 2022-08-27:
We've got a handful of positive fixes in v2.4. I'm going to close this for now but please open a new issue if you still see problems. We've got a few more improvements to make to logging in v2.5 as well.

closed 2022-08-27 by nseidle