sparkfun/SparkFun_RTK_Firmware #121: Memory Loss: Rover Fix -> Display Setup -> Rover Fix

LeeLeahy2, 2022-05-14:
### Subject of the issue
Pushing the setup button while in Rover Fix and then walking the setup down to Rover reduces the free space in the heap by just over 2K bytes.

### Your workbench
* What version of RTK Surveyor firmware are you running?   main (6126b9151fb518471f35d9e838dd433a61b5f03c)
* Are you connected to the device over Bluetooth? What app?   No
* Are you transmitting NTRIP back to the device?   No
* How is everything being powered?   Battery / USB cable
* Are there any additional details that may help us help you?

### Steps to reproduce
On an RTK Express system:

1. Power on the system in Rover mode
2. Using the serial config, enable heap debugging
3. Allow Rover to get a fix
4. Press and release the Setup button
5. Press and release the setup button one or more times until the Rover entry is selected
6. Allow the RTK to transition back into Rover mode

### Expected behavior
The free heap space should not decrease

### Actual behavior
The free heap space is reduced by about 2K bytes.  The output below shows the free space upon state entry, the range of free heap values, and the free space upon state exit followed by the next state:

`    |
    |  Free Heap bytes
    |  79868
    |  79776 <= bytes <= 79868
    |  79776
    V
Display Setup
    |
    |  Free Heap bytes
    |  79776
    V
WiFi Config Not Started
    |
    |  Free Heap bytes
    |  79776
    |  79776 <= bytes <= 166924
    |  166924
    V
WiFi Config
    |
    |  Free Heap bytes
    |  166924
    |  75060 <= bytes <= 166924
    |  75060
    V
Display Setup
    |
    |  Free Heap bytes
    |  75060
    V
WiFi Config Not Started
    |
    |  Free Heap bytes
    |  75060
    V
WiFi Config
    |
    |  Free Heap bytes
    |  75060
    |  72452 <= bytes <= 75060
    |  72452
    V
Display Setup
    |
[... 465 lines cut ...]
    77368
    75384

----------------------------------------------------------------------
State: Base - Not Started
Free heap bytes upon entry
    73736
    79384
    79388
    79392
    79400
    79392
    79388

----------------------------------------------------------------------
State: Base-Temp - Settle
Free heap bytes upon entry
    166904
    166560
    166560
    166560
    166560
    166560
    166560
`

nseidle, 2022-05-14:
I appreciate the reporting. Knowing my programming skills, a memory leak is not surprising!

I can confirm this: 74124 to 72132 when re-starting Rover mode. However, removing WiFi and BT from compilation (commenting out COMPILE_BT, etc) the stack is maintained: 253576 before, 253576 after. So my first inclination is to point the smoking gun at the built in BT library. 

In my experience, the BT and WiFi libs leak a bit when switching. The built-in 'WiFiBlueToothSwitch' example (ESP32 core) is a good starting spot for switching between WiFi and BT. I had a sketch long ago that also printed the heap. I will try to find it or rebuild it.

We can certainly start by make the startBluetooth/stopBluetooth functions re-entrant friendly.

nseidle, 2022-05-14:
Ah! The [switch sketch](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/main/Firmware/Test%20Sketches/WiFiBluetoothSwitch/WiFiBluetoothSwitch.ino) is still there in the firmware folder.

nseidle, 2022-05-14:
Initial findings:

* It looks like the ESP32 has fixed the BT.end() / BT.begin again bug that was causing crashing and required the use of the customBTstop() code. We can likely excise that. 
* Running the switch sketch, I am not seeing a memory leak with starting/stopping BT. That too must have been fixed over the past two years. 235612 no BT, 148280 with BT, 235612 after stopping BT again (note: it looks like it takes a few 100ms for the FreeRTOS to clean up all the tasks during BT.end).

nseidle, 2022-05-15:
Have a look at this [commit](https://github.com/sparkfun/SparkFun_RTK_Firmware/commit/1e2fb3257bc1af072106c553e724910d03072593). Re-starting rover, there is no more memory leak. I've cleared out all the old fixes and code that is no longer needed (due to core fixes).

closed 2022-06-01 by nseidle