sparkfun/SparkFun_RTK_Firmware #124: Log Time in RTK Surveyor

IngridBerru, 2022-05-16:
Good Morning,
I have the Sparkfun RTK surveyor ( ZED-F9P) and I updated it to the last firmware version (V 1.12 and HPG1.32), however I have two limitations which are the following:
*Minimum Measurement rate is 0.1, however, in the project we require to have measurement rates less than that for example 0.05 Can we change the configuration to have a measurement rate less than 0.1?
* when was configuring the equipment It is said that the log time and max log length must be between 1 to 2880, however, in the project we need more time for example 6 months of continuous data, Is it possible to obtain the log time require and How can we change it?
Thank you so much for your support.

nseidle, 2022-05-16:
Hi - I did not know there was a HPG firmware version 1.32. I think I'm still on v1.3. Can you provide a link?

> Minimum Measurement rate is 0.1

[image]

The current firmware supports 10Hz. Max nav rate for RTK is 7Hz and RAW is 15Hz (in general). Are you prepared to drop the other constellations to get to 20Hz?

> When I was configuring the equipment It is said that the log time and max log length must be between 1 to 2880, 

This was increased to [2 years](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/main/Firmware/RTK_Surveyor/menuMessages.ino#L49) in v1.11 2880 minutes (48 hours) is the max time between sequentially created logs. I recommend you set the logging time to 6 months, and the max log length to 24 hours. This is to prevent accidental corruption of an entire 6 month log.

IngridBerru, 2022-05-18:
The link to the HPG 1.32 firmware is below:
[(https://www.u-blox.com/search/?pages%5Bfile%5D=3&query=HPG%25201.32)]

Regarding the minimum Measurement rate, I was talking about Hz, in our project, we need 0.06Hz, which is about 4 measurements per minute, so we're wondering if we can reach that frequency. 

Also, Do you recommend changing the rate for RTK, PVT, and Raw at the same rate of 0.06 Hz, or the minimum that I could reach that was 0.1 Hz?
Thank you so much for your answer

nseidle, 2022-05-19:
> HPG 1.32 firmware

Thank you for letting us know. We'll add this support to the firmware. It should work fine as it is.

> 4 measurements per minute

Ah, ok. Yes, we should be able to support that. I will double check that the ZED supports that.

> Do you recommend changing the rate for RTK, PVT, and Raw at the same rate of 0.06 Hz, or the minimum that I could reach that was 0.1 Hz?

I don't fully understand what you are trying to achieve. Are you needing to log these values once per 4 minutes? With the current firmware you can set the device measurement rate to 1Hz, then set the individual message (NMEA, RAWX, etc) rates to 10. This will cause the unit to log the messages once per 10 seconds. Decreasing the measurement rate or increasing the message rate should enable you to get the logging rate you need.

IngridBerru, 2022-05-20:
Thank you so much @nseidle, We need to have 4 measurements per minute which is approximately 0.06 Hz, so what I was asking is to have a homogeneous record if I can set the device measurement rate to 0.06 Hz and then set the individual message (NMEA, RAWX, etc) rates to 16.6 seconds.

nseidle, 2022-05-20:
> 4 measurements per minute

Ah, I understand now. This should be easy to do. Use the 'Set measurement rate in seconds between measurements:' menu option at set it to your desired time.

[image]

Below are the default NMEA message rates:

[image]

Using these default message rates the NMEA GGA sentence will appear in the log every 16.6 seconds. The NMEA GSV sentence will appear every 16.6*4 or 66.4seconds.

As an aside, where did you originally read 'Minimum Measurement rate is 0.1'?

closed 2022-05-20 by nseidle

IngridBerru, 2022-05-20:
Hello @nseidle , I didn't read about the 0.1HZ is the Minimum Measurement rate, I assume it because when I was changing the configuration using WIFI (192.168.4.1), It didn't allow me to save the configuration and an error message appeared, so I tried different values and when I entered 0.1 HZ it allowed to save it; however, when I used the Tera Term software I didn't have the same issue.

Something similar happened with the log time I changed the " Max Log Time" using  Tera Term software for a period of 1 month, but when I tried the same using a WIFI connection (192.168.4.1), It didn't allow me.