sparkfun/SparkFun_RTK_Firmware #131: All serial configuration state lost upon reset

LeeLeahy2, 2022-05-16:
### Subject of the issue
All serial configuration state is lost upon reset.  A reset after setting up WiFi access in Rover mode and enabling heap monitoring causes these configuration values to be lost.  Reset is being done using the 'r' selection in the debug menu.

### Your workbench
* What version of RTK Surveyor firmware are you running?   release_candidate (9b68e43f3e16bf67ed1099e530522d2fb47cf223)
* Are you connected to the device over Bluetooth? What app?   No
* Are you transmitting NTRIP back to the device?   No
* How is everything being powered?  Battery / USB cable
* Are there any additional details that may help us help you?  No

### Steps to reproduce
1. Power on the RTK express system
2. Enter the serial config menu by pressing Enter
3. Using serial config, select 1 in the main menu
4. Select 6 and specify the WiFi network
5. Select 7 and specify the WiFi password
6. Select 14 to disable
7. Exit back to the main menu
8. Select 's' for the System menu
9. Select 'd' for the Debug menu
10. Select 2 to enable the heap monitoring
11. Exit all of the serial config menus
12. Verify the RTK is connected to WiFi and displaying the heap free space
13. Press Enter to enter the serial configuration menu
14. Press 's' to enter the System menu
15. Press 'd' to enter the Debug menu
16. Press 'r' to reset the RTK system
17. Allow the RTK system to restart
18. The RTK is no longer connected to WiFi
19. The RTK is no longer printing out the heap free space

### Expected behavior
Previous versions of the RTK firmware would keep the settings across reset.

### Actual behavior
Configuration settings are being lost across reset.

nseidle, 2022-05-16:
recordSystemSettings() doesn't get called until you exit the serial menu. I'll add it before a reset.

LeeLeahy2, 2022-05-16:
I don't know if the state is not getting saved or reloaded.  But in either case, the RTK Express does not have the settings following reset.

nseidle, 2022-05-16:
This is already fixed in [this commit](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/131). Give it a try and re-open if you still have this problem.

LeeLeahy2, 2022-05-17:
I did sync my sources, built them and downloaded the image to the RTK Express, prior to my previous comment.  The data is not getting saved, with or without exiting from the serial config back to normal execution.  Once the reset happens, all configuration data is lost!

LeeLeahy2, 2022-05-17:
Configuration:
RTK Express system
16 MB SPI flash
3 MB App
9 MB FATFS
No microSD card

LittleFS is not getting initialized in this configuration, See attached [log file](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/8710863/20220517_0904.txt)!  
Reducing to 8 MB flash, with 3MB App and 1.5MB FATFS seems to work.

I recommend removing the recordSettings call before debug reset.  If you disable logging to the microSD card and remove the card and then use Serial config to reset the system (ENTER, s, d, r) the system hangs attempting to write to the microSD card.

nseidle, 2022-05-17:
Your log is good but is missing the initial LittleFS mount. Did it fail from the power on? If yes, then I suspect you need to do a s-d-f to reformat the LittleFS SPIFFS partition. 

When I get my unit into a funky state, the reset to factory defaults is a good way to return to known. Factory Reset includes LittleFS format.

> I recommend removing the recordSettings call before debug reset.

But I like it :) It makes it so that I don't have to exit the serial menu, just to re-enter then do a s-d-r reset.

> If you disable logging to the microSD card and remove the card and then use Serial config to reset the system (ENTER, s, d, r) the system hangs attempting to write to the microSD card.

The system hangs because we don't support hot swap removal. We could add an 'Dismount SD' menu item if needed, but that's another issue to be opened (I swear you're going for a record! Keep them coming!).

I'm reopening this as it seems you're still having the original issue.

LeeLeahy2, 2022-05-18:
The initialization of LittleFS near line 344 in Begin.ino indicates that it will format LittleFS on the failure path.  The initial mount of LittleFS is failing.  [Erasing](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/8718891/20220518_0731.txt) LittleFS also fails.  [Resetting](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/8718916/20220518_0750.txt) to factory settings also fails to initialize LittleFS!

nseidle, 2022-05-18:
Ah - it's all sorts of messed up. Please see the most recent [compilation instructions](https://github.com/sparkfun/SparkFun_RTK_Firmware/tree/release_candidate/Firmware). I suspect I failed to warn you that the partitions changed. Key points:

* Change the Parition table. Replace 'C:\Users\[user name]\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\tools\partitions\app3M_fat9M_16MB.csv' with the app3M_fat9M_16MB.csv file found in this folder. This will increase the program partition from a maximum of 1.9MB to 3MB.
* Set the core settings: The 'Partition Scheme' must be set to '16M Flash (3MB APP/9MB FATFS)'. This will use the 'app3M_fat9M_16MB.csv' updated partition table.

LeeLeahy2, 2022-05-18:
Much better, LittleFS is now working

nseidle, 2022-05-18:
Yay! Thanks for reporting back.

commit 4dfc4e5b (2022-05-16): Fix #131 - save settings before debug reset
closed 2022-05-18 by nseidle