sparkfun/SparkFun_RTK_Firmware #132: Data not freed when enabling Rover Client NTRIP

LeeLeahy2, 2022-05-16:
### Subject of the issue
About 20K of data is not being freed when enabling Rover Client NTRIP mode.

### Your workbench
* What version of RTK Surveyor firmware are you running?   release_candidate (9b68e43f3e16bf67ed1099e530522d2fb47cf223)
* Are you connected to the device over Bluetooth? What app?   No
* Are you transmitting NTRIP back to the device?   No
* How is everything being powered?  Battery / USB cable
* Are there any additional details that may help us help you?  No

### Steps to reproduce
1. Power on the RTK Express system
2. Allow the system enter Rover mode and obtain a fix
3. Press Enter to enter the serial configuration mode
4. Enter '1' to enter the menu to specify the Rover's NTRIP Client
5. Enter '6' to enter the WiFi network name
6. Enter '7' to enter the WiFi password
7. Enter '14' to disable
8. Exit all of the menus
9. Allow the RTK Express to make its WiFi connection and enter the RTK float state
10. Press and release the Setup button to enter Setup mode
11. Press and release the Setup button one or more times to select Rover mode
12. Allow the RTK Express to exit the Setup mode

### Expected behavior
The RTK system should free up all memory during the "Rover - Not Started" state

### Actual behavior
When enabling the Rover NTRIP Client, the RTK system does not release approximately 20K of heap memory.  This memory does get released by entering Setup mode and exiting in Rover mode.

See [junk.txt](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/8702776/junk.txt)

pyrog, 2022-05-17:
Several test of the NTRIP client show that it is not reliable. The RTK led is off after several minutes.
This is probably related to memory leak ??

See #134

nseidle, 2022-08-18:
Testing with v2.4 firmware:

Rover, No NTRIP Client: FreeHeap: 90344 / HeapLowestPoint: 72100 / LargestBlock: 73716
Rover, Connected to NTRIP Caster: FreeHeap: 51104 / HeapLowestPoint: 34520 / LargestBlock: 34804
Rover, No NTRIP Client: FreeHeap: 74376 / HeapLowestPoint: 27292 / LargestBlock: 36852
Rover, Connected to NTRIP Caster: FreeHeap: 51240 / HeapLowestPoint: 27292 / LargestBlock: 34804

We can see the initial WiFi connection takes the heap down. Exiting the WiFi doesn't fully release it. But re-enabling WiFi shows the same heap level (51k). It's an initial heap hit but not a running leak IMO. I'm closing. If someone wants to dig into the ESP32 WiFi core, I welcome the PR.

closed 2022-08-18 by nseidle