sparkfun/SparkFun_RTK_Firmware #135: RTK Surveyor on in Base Mode- Bluetooth

IngridBerru, 2022-05-19:
Good Afternoon
I wanted to see the position of the Base by SW Maps however when the RTK Surveyor was on Base Mode there wasn't any blue light and when I opened the SW maps It said: "It couldn't connect". Do you know what could be happening?

nseidle, 2022-05-20:
Depends on whether you are in fixed or temporary base mode.

Bluetooth is turned off during Fixed Base mode. We do not expect users to need or want to obtain lat/lon from a device while it is in Base Mode since they have previously entered their known fixed coordinates. 

Bluetooth should be available during a temporary base however it will not transmit location data while the device is settling, or while the survey-in process is happening. Once the device has completed survey-in it should begin to transmit the fixed lat/lon in case it is needed.

If there is a use case where you need to see fix data during base mode, please let me know.

IngridBerru, 2022-05-20:
Thank @nseidle  for your answer It clarifies a lot. I set the base as a temporary using the survey-in mode and it started to transmit the RCTM corrections to the rovers and I was curious to see the coordinates of the base, but when I used the  SW Maps app It couldn't connect.

nseidle, 2022-05-20:
Happy to help!

closed 2022-05-20 by nseidle