sparkfun/SparkFun_RTK_Firmware #137: WiFi NTRIP client is not documented

gdt, 2022-05-19:
I am trying to understand what is possible and what is not, at the moment, and while the hookup guide is very useful, it isn't really a manual or a description of what is possible.   Perhaps it can just be extended; I don't mean to suggest a separate manual.

Perhaps there should be a firmware manual as that's 99% the same with multiple products, a hookup guide for each about the connectors, and tutorials for setup in various ways.

firmware 1.12, but really just looking at this repo and the hookup guide (reading the Facet one because I know updates are being made  there).

Specific information that is missing is:
 - That this ability exists.
 - WiFi "PW" is obviously WPA2 :-) but it's not actually obvious.
 - Does configuring NTRIP client support the same options that the base NTRIP client supports?
 - Are there constraints on simultaneous operation of WiFi NTRIP client and BT?  (The computer scientist would expect not, even if the firmware engineer would worry about RAM usage.)

gdt, 2022-05-19:
Sorry, dup of #109.  I thought I had read the issue list...

closed 2022-05-19 by gdt

nseidle, 2022-05-20:
No worries. This is good feedback. I'll see about splitting things up a bit. 

> Are there constraints on simultaneous operation of WiFi NTRIP client and BT?

There is no physical constraint, only RAM constraints. We've successfully done this but the unit becomes unstable if any other features are activated so we're shying away from supporting it for now.