sparkfun/SparkFun_RTK_Firmware #138: OTA firmware update fails

LeeLeahy2, 2022-05-20:
### Subject of the issue
The Over The Air (OTA) firmware update fails.  On an RTK Express system running v1.10 firmware, entering WiFi Config mode and then initiating a firmware upload of RTK_Surveyor_Firmware.bin displays the message "OTA could not begin" on the browser and then reboots due to a failed assertion.

### Your workbench
* What version of RTK Surveyor firmware are you running?   v1.10
* Are you connected to the device over Bluetooth? What app?   No, connected via WiFi to 192.168.4.1
* Are you transmitting NTRIP back to the device?   No
* How is everything being powered?  Battery / USB cable
* Are there any additional details that may help us help you?  See attached [log file](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/8743421/20220520_0846.txt).

### Steps to reproduce
1. Power on the RTK Express system
2. Load firmware v1.10
3. Power on the RTK Express system
4. Press and release the Setup button to enter Setup mode
5. Press and release the Setup button multiple times until WiFi Config is selected
6. Wait for the RTK Express to enter WiFi config mode
7. On a PC or MAC, select the WiFi network "RTK Config"
8. With a browser, connect to http://192.168.4.1/
9. Click on the "System Configuration" button near the bottom of the screen
10. Click on the "Choose File" button
11. Using the dialog box that pops up select the "RTK_Surveyor_Firmware.bin" file
12. Click the upload button

### Expected behavior
The system should fail gracefully, displaying a message on the browser and then continue operation in WiFi Config mode.

### Actual behavior
The system resets due to a failed assertion.

I realize that this is old firmware and that the fix will not be made to the old firmware.  However a fix should be applied to any new firmware to enable the download to fail gracefully back to WiFi Config mode.

What I was trying to test was the upgrade path from old firmware to new firmware.  There have been several changes to the partition structure over the past couple of months.  As I understand things, the old APP partition was 1.9MB and the when the current image is greater than 1.9MB I would expect the firmware download to fail.

I believe there needs to be a small (< 1.9MB) intermediate firmware image built using the new partition structure.  This image gets downloaded to the device before any new firmware images are downloaded.

Eric-FR, 2022-05-21:
I think the issue also exists with Sreial Config.

I copied the new firmware on the µSD card, put the µSD card in the RTK Express, connected with USB cable to desktop and connected with arduino serial monitor. The firmware is available in the list but the upgrade fails with message:
`Updage begin failed. Not enough partition space available.`

[image]

pyrog, 2022-06-29:
Same issue from v2.0-May 26 2022  to [v2.3RC-June 28 2022](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/download/v2.3RC/RTK_Surveyor_Firmware_v2_3RC-June28.bin).
Display `OTA could not begin`.

Tested with RTK Surveyor and RTK Express.

pyrog, 2022-06-29:
Same issue from USB
```
SparkFun RTK Surveyor v2.0-May 26 2022
Menu: Main Menu
f) Firmware upgrade
f

Menu: Update Firmware Menu
1) Load RTK_Surveyor_Firmware_v2_0.bin
2) Load RTK_Surveyor_Firmware_v2_3RC-June28.bin
2
Loading RTK_Surveyor_Firmware_v2_3RC-June28.bin

Update begin failed. Not enough partition space available.
```

nseidle, 2022-06-29:
@Eric-FR and @pyrog - This is a [known issue](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/151).

**Note:** If you are currently using version 1.x firmware you will need to load v2.x firmware using a USB connection (SD and WiFi are not supported). We have created a Windows GUI to ease this process. Please see [Updating Firmware from GUI](https://sparkfun.github.io/SparkFun_RTK_Firmware/firmware_update/#updating-firmware-from-gui) for a step by step process. Once you've updated to v2.x firmware, you will be able to load future firmware using any method (SD, WiFi, USB, or GUI).

Let me know if the GUI does not get you fixed.

The issue that Lee opened is with regards to better handling the error itself via the WiFi AP page, not that the update fails. We know the update fails due to partition sizes. The AP config page should properly guide the user to use the GUI.

Eric-FR, 2022-06-29:
Done with the GUI tool. Upgraded to 2.2 ans running.

nseidle, 2022-08-18:
This error is limited to users who have v1.x of the firmware on their unit. Once they've upgraded to 2.x, their partition files will be corrected and OTA is possible again, and they shouldn't hit this error. If someone does, please open a new issue.

closed 2022-08-18 by nseidle