sparkfun/SparkFun_RTK_Firmware #144: Intermittent RCTM Corrections

IngridBerru, 2022-05-27:
Hello,
I set a Base using survey-in and RTK Surveyor ( Firmware RTK v2.0) and I configurated to send RCTM corrections to two rovers (RTK Surveyor Express Pluss), however every 5-10 minutes the rovers didn't recognize the RCTM corrections and the rovers changed their fixed mode from 3D to Float and then Fixed for less than 1 minute. This happened during the whole testing ( Approximately 1.5 hrs). I was wondering if this is usual or what can it be happening?.

The Base frequency and Rovers are set to 0.06 Hz.

Thank you very much for your support.

nseidle, 2022-06-02:
Sorry for the delay in responding. RTK float/fix is maintained with a regular stream (1Hz) of RTCM corrections. If your base is transmitting at 0.06Hz then I am not too surprised the Rover has RTK lock issues. It needs to be fed a regular stream of RTCM to obtain RTK lock. 

It usually only takes a few seconds to get RTK float then lock, and I've seen RTK maintained for 15 to 20s when the RTCM feed is blocked/stopped, but the Rover eventually gives up and drops back to normal, non-RTK fixes. The Rover is not expected to obtain RTK with a single RTCM packet reception. Perhaps everything is working but your radio is corrupting one of the 16.6s packets every 5-10 minutes. I don't know how resilient your radios are to interference.

Is there a reason why your base is set to 0.06Hz? Do you see the same problems if the base is set to 1Hz?

IngridBerru, 2022-06-03:
Hello @nseidle ,
The base and rovers are set to 0.06Hz because we're thinking to implement the Sparkfun units in a Landslide and monitoring the movement for a long period of time ( 6 months or more), hence we don't need to have a higher frequency because the movement is slow and to have enough space in the SD card to store data for the required period of time.

We're using the "Telemetry Radio 915 mhz 100 mW"; I'll be changing the base to 1 Hz to see if we have the same problems, in case I'll experience the same issues do you know what we can do to fix it?

Thank you so much for the previous reply.

nseidle, 2022-06-09:
> We're using the "Telemetry Radio 915 mhz 100 mW"

Those work well but are not impervious to data corruption.

> I'll be changing the base to 1 Hz to see if we have the same problems, in case I'll experience the same issues do you know what we can do to fix it?

Let's cross that bridge when we get there. Please give the 1Hz a try and report back if you have success or not.

nseidle, 2022-08-01:
Closing due to inactivity. Please open an issue if you still have problems.

closed 2022-08-01 by nseidle