sparkfun/SparkFun_RTK_Firmware #145: SDCARD Semaphore failed to Yield

IngridBerru, 2022-05-27:
Hello,
I updated the base and rover to the last firmware update (v.2.0) and when I connected the base to look at the generation rate using Terra Term. A message appeared (See Below) I was wondering if that issue will cause an error during the PPP of the base or what it means. To clarify this only happens in the base.

[image]

Thank you so much for your help.

nseidle, 2022-05-27:
Thanks for reporting. The semaphore failing to yield is a warning only and snuck in at the last minute. It's been changed to a debug message within v2.0 firmware [here](https://github.com/sparkfun/SparkFun_RTK_Firmware/commit/8127a7e2ca2fa383ef84fb6a0de583890824ddaa). Please re-load v2.0 binary and you shouldn't see those anymore.

It is a warning only. The buffers are large enough to continue logging at normal (4Hz) rates even with a temporarily blocked semaphore.

closed 2022-05-27 by nseidle

LeeLeahy2, 2022-06-12:
The error displayed is an error where data is not being properly written to the log file.