sparkfun/SparkFun_RTK_Firmware #149: L-Band is a confusing name

gdt, 2022-05-29:
At https://sparkfun.github.io/SparkFun_RTK_Firmware/configure_lband/ there is talk of "L band".  This doesn't really make sense as GPS signals are within the L band, as are the other constellations.  Reading more carefully it becomes clear that this is all about the PointPerfect service from u-blox, which happens to also operate in the L band.   I would therefore suggest renaming the menu and doc section to be about PointPerfect rather than the band designation.   And if you can't use that word (but yet you do) then how about "Satellite SSR RTK Data".   (I avoided using "corrections" because RTCM3 data for RTK is not corrections.)

nseidle, 2022-05-31:
Agreed. I've changed that menu and much of the references to include some form of 'PointPerfect' as opposed to L-Band. I'll get a release candidate posted shortly.

nseidle, 2022-05-31:
RC is available [here](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.1). We'll get the docs updated as well.

nseidle, 2022-06-02:
[Docs](https://sparkfun.github.io/SparkFun_RTK_Firmware/configure_pointperfect/) are updated. Please open a new issue if you see anything else.

closed 2022-06-02 by nseidle