sparkfun/SparkFun_RTK_Firmware #151: Failed to update Firmware V.2.1

IngridBerru, 2022-06-01:
Hello,
Currently, the Sparkfuns ( Rover and Base) have the firmware version v.2.0 ; and I want to update to the new version 2.1 however I experienced the following problems:

*Using the GUI, the device doesn't reset and power down, as a result, the firmware is not updated and the device kept frozen, however, the messages on the GUI say that the device is resetting.
[image]

*Using Wifi, also It doesn't allow it to update, for example at the end of the updating process the following message appears "OTA could no begin".
[image]

Best regards,

nseidle, 2022-06-02:
The GUI failure is due to a bad partition file. We'll update the GUI shortly.

tonycanike, 2022-06-02:
I tried to update from V2.0 to V2.1 using the Windows GUI and also via the SD update path.  Was unsuccessful.  Here's the serial log when I tried using the SD card method:

```
SparkFun RTK Facet v2.0-May 26 2022
** Bluetooth broadcasting as: Facet Rover-4DAE **
Menu: Main Menu
1) Configure GNSS Receiver
2) Configure GNSS Messages
3) Configure Base
4) Configure Ports
5) Configure Logging
p) Configure Profiles
s) System Status
f) Firmware upgrade
x) Exit

Menu: Update Firmware Menu
1) Load RTK_Surveyor_Firmware_v1_10.bin
2) Load RTK_Surveyor_Firmware_v1_11.bin
3) Load RTK_Surveyor_Firmware_v1_12.bin
4) Load RTK_Surveyor_Firmware_v2_0.bin
5) Load RTK_Surveyor_Firmware_v2_1.bin
x) Exit
5
Loading RTK_Surveyor_Firmware_v2_1.bin
Update begin failed. Not enough partition space available.

Menu: Update Firmware Menu
1) Load RTK_Surveyor_Firmware_v1_10.bin
2) Load RTK_Surveyor_Firmware_v1_11.bin
3) Load RTK_Surveyor_Firmware_v1_12.bin
4) Load RTK_Surveyor_Firmware_v2_0.bin
5) Load RTK_Surveyor_Firmware_v2_1.bin
x) Exit

```

nseidle, 2022-06-02:
Sorry the GUI is currently down. We'll have it fixed shortly. 

@tonycanike - Your partitions are set to 2MB preventing v2.1 to be loaded. The GUI will eventually fix partition issues, but in the between times, do you feel like trying the [CLI method](https://sparkfun.github.io/SparkFun_RTK_Firmware/firmware_update/#updating-firmware-from-cli)?

Once we get the partitions sorted, SD and WiFi update paths work again:

[image]

tonycanike, 2022-06-02:
Thanks @nseidle that worked perfectly.

PaulZC, 2022-06-02:
Hi @IngridBerru / @tonycanike ,

Version 1.1 of the firmware uploader executable is now available. You can download it on its own using the [download button on this page](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/main/Uploader_GUI/Windows_exe/RTK_Firmware_Uploader_GUI.exe). Or you can download the whole repo using the usual methods. The exe contains the corrected partition file. Please give it a try and let us know if you see any issues.

Thank you,
Paul

PaulZC, 2022-06-02:
PS. If - like me - you have a very old RTK Surveyor - it may contain a 4MB version of the ESP32. If you have one too, I'm afraid you're restricted to [v1.0 of the GUI](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/main/Uploader_GUI/Windows_exe/Previous_Version/RTK_Firmware_Uploader_GUI.exe) and [v2.0 of the firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/main/Binaries/RTK_Surveyor_Firmware_v2_0.bin). At least for now...

You can tell which version you have by looking at the diagnostic messages from the upgrade:

[image]

tonycanike, 2022-06-03:
> Hi @IngridBerru / @tonycanike ,
> 
> 
> Version 1.1 of the firmware uploader executable is now available. ... The exe contains the corrected partition file. Please give it a try and let us know if you see any issues.
> 

Thanks @PaulZC! I had already upgraded to 2.1 via CLI per @nseidle's suggestion.

nseidle, 2022-06-03:
Excellent! Glad to hear it.

closed 2022-06-28 by nseidle