sparkfun/SparkFun_RTK_Firmware #154: Repository branch structure is confusing and not documented

gdt, 2022-06-02:
Looking at README.md, I can't figure out a few things that README.md should describe.  For background, I've been using git since 2010 and writing software for much longet than that.

  - How would I build the firmware?   (That can be in BUILDING.md which I would have found.)   Presumably there are some prereq packages, and some build tools.   I'm sure I could figure it out, but a project should explain this.

  - What are the branches for?  I see main and release_candidate.   I know there are styles with main and develop, and what I consider the standard approach with new code on main and release branches.  RC is usually a tag, not a branch.   Right now I see a 2.1_RC binary in release_candidate, but a 2.1 tag on main, where there is no binary.   README should explain which branch is for what and what the cross-merging plan is.

nseidle, 2022-06-02:
> How would I build the firmware?

https://sparkfun.github.io/SparkFun_RTK_Firmware/firmware_update/#compiling-from-source

I'm happy to drop a pointer from BUILDING.md to this section. While we have *a lot* of repos, it's rare that we have firmware/software that requires this level of instruction.

> What are the branches for?

Again, the vast majority of our repos don't have this level of firmware complexity so we're not accustom to formal branch structures. I welcome recommendations and codification where needed. In general, main is the latest release, release_candidate is the user-beware-sandbox. Once the RC reaches a level of testing and feedback, we 'release' and merge to main.

gdt, 2022-06-02:
Thanks, I'll submit a PR to explain the branches.  What you are doing seems fine; it is just using unusual names so should be stated.  I'll also add in  apointer to the build section.  (That needs some help, but the big point is making it easily findable from README; I didn't expect build instructions to be included in the "how to use this program" manual.  A line or two entirely resolves this.)

gdt, 2022-06-10:
It's still confusing to have "develop" be called release_candidate, when in general it is not, but at least now it is documented.

closed 2022-06-10 by gdt