sparkfun/SparkFun_RTK_Firmware #156: Firmware update instructions assume Windows

gdt, 2022-06-02:
At https://sparkfun.github.io/SparkFun_RTK_Firmware/firmware_update/#updating-firmware-from-gui the firmware update instructinos say to run something ending in .exe.  It is obvious (but perhaps not true) that this shouldn't be done on non-WIndows, such as macOS, GNU/LInux or NetBSD.   It's also obvious that given that esptool is in python that a python updater should work, and that maybe this is that -- but you just don't run foo.exe on non-Windows.

The CLI section talks about finding device path from windows (and I know how to find /dev/dtyU0 or /dev/complicated-mac-name but I bet some don't), and then to run esptool.exe.   Again probably this is the python tool wrapped somehow, but rule  number 1 of firmware updating is do it carefully, do it right, and do not guess at anything lest you brick something.

I believe that the actual situation is almost certainly very ok, and a few lines of clue would go a long way to making this easy for people not using WIndows.

nseidle, 2022-06-02:
Not to poke too hard, but aren't you the contributor who recommended we [move the docs](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/137) to a single, user editable, documentation system? :) Well, ok, you didn't state user editable; that was my goal.

!Yes! Please create PRs with recommended edits.

To the root of the issue, I am primarily a windows user and @LeeLeahy2 is always kindly pushing me from the Linux side so I welcome the push for additional OS support.

The GUI executable is very Windows. Would the manual section be better simply titled 'Updating Firmware from Windows GUI'? I assumed (perhaps lazily) the python based espupdate tool would be sufficient for macOS and Linux, but correct me if I'm wrong.

(I appreciate the separate issues btw)

LeeLeahy2, 2022-07-04:
Do the instructions for Ubuntu (Linux) help?

https://sparkfun.github.io/SparkFun_RTK_Firmware/firmware_update/#ubuntu-2004

closed 2022-07-05 by nseidle