sparkfun/SparkFun_RTK_Firmware #159: v2.1 tag is in an odd place

gdt, 2022-06-02:
The v2.1 tag is on main on a commit that predates 2.1.   In the README PR, I documented that release tags are on main - but I may have gotten that wong.

nseidle, 2022-06-03:
No this is my fault. I create the firmware version tag but mark it as a pre release. Once we're do a release, we remove the pre release mark but github doesn't do an update of the source at that point in time of main. I'm doing something wrong. I may create tags with RC in the future and then just abandon the RC tag in place of a new release tag.

gdt, 2022-06-03:
My advice, for what it's worth is:
  - Do not create a release tag except as the release.
  - maybe create rc tags, but maybe just ask people to test (a malinglist would be great for this)
  - Until the binary problem is solved, not that each binary commited takes up space forever in the .git subdir; this repo is at 500 MB already.
I think you can delete a release and recreate it.