sparkfun/SparkFun_RTK_Firmware #168: Not able to get into STATE_ROVER_RTK_FLOAT or STATE_ROVER_RTK_FIX

LeeLeahy2, 2022-06-13:
### Subject of the issue
The RTK Express and RTK Facet L-Band do not seem to transition into STATE_ROVER_RTK_FLOAT or STATE_ROVER_RTK_FIX.

### Your workbench
* What version of RTK Surveyor firmware are you running? Multiple versions on both RTK systems
* Are you connected to the device over Bluetooth? NO, using WiFi
* How is everything being powered?  Battery and USB-C cable
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please include a copy of your settings.txt file used on the unit.

### Expected behavior
With the NTRIP client enabled with valid mount point and password and valid WiFi SSID and password, I expect the Rover to enter into either *_FLOAT or *_FIX with a couple of minutes.

### Actual behavior
The RTK devices never get there!

nseidle, 2022-06-13:
(Communicated with @LeeLeahy2 off github) - I suspect this is due to a faulty correction source. Please setup a temporary base, feed the corrections to RTK2go using the NTRIP Server, and then attempt rover with NTRIP client.

nseidle, 2022-07-05:
Closing. Permanent base was the issue.