sparkfun/SparkFun_RTK_Firmware #171: How do I set the time zone?

LeeLeahy2, 2022-06-13:
### How do I set the time zone?
The wall clock time is 1:40 PM but both the RTK Express and the RTK Facet L-Band believe that it is 11:40 AM.  How do I set the time zone so that the time is correct?

### Your workbench
* What version of RTK Surveyor firmware are you running?  0cf9cf92f1cd4ddd6bbfc80ee611bf4d886de3d9 +
* Are you connected to the device over Bluetooth?  No, connected via WiFi
* How is everything being powered?  Battery + USB-C cable
* Are there any additional details that may help us help you?  + = dynamic-sd-card + sd-card-server + marks-file

### Steps to reproduce
1. Power off the RTK
2. Remove the microSD card
3. Using another system, delete the files on the microSD card
4. Insert the microSD card into the RTK
5. Power on the RTK
6. Switch to Rover mode
7. Enable the NTRIP client (WiFi)
8. Look at the dates and times on the files on the SD card
9. Compare with the local time

### Expected behavior
The log file time should match the wall clock time within a few seconds

### Actual behavior
In my case the time is off by 2 hours

gdt, 2022-06-14:
I would expect times to all be UTC.

nseidle, 2022-06-14:
This is a duplicate of #110. All time stamps are UTC. Currently there's no feature to set timestamps to a different time. I'm open to a feature request, timestamp mods are easy enough.

closed 2022-06-14 by nseidle

LeeLeahy2, 2022-06-14:
Please look at the file times on your system  They are most likely incorrect, even if they are specified as GMT (UTC).

I am pulling real time clock (rtc) time.  It looks like the file times are also based on the real time clock (rtc).

My time zone is GMT - 10 hours.  The comments in the code indicate that rtc.getHours() returns values 0 - 23.  So for my example if the wall clock reads 13:40 (wall clock = GMT - 10 hours) then GMT should be 23:40 (GMT = wall clock + 10 hours).  However the actual value from rtc.getHours() is 11.  So either rtc.getHours() is returning values 1 - 12 with AM & PM begin returned by another function or the real time clock was set improperly using the values from the GNSS.  In the latter case 
the AM / PM value is being ignored when setting the real time clock, example 23:40 - 12:00 = 11:40 AM.

nseidle, 2022-06-14:
I recommend opening a new issue as this started as a feature request and sounds like a bug.