sparkfun/SparkFun_RTK_Firmware #207: Firmware update from Linux fails: esp32 boot loop

eshattow, 2022-07-04:
SparkFun RTK Surveyor. With RTK_Surveyor_Firmware_v2_0.bin loaded a firmware upgrade of RTK_Surveyor_Firmware_v2_2.bin from the SD card fails because there is no room on the partition. Note the startup sequence for v2.0:
```
[06:33:09:917] rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)␍␊
[06:33:09:922] configsip: 0, SPIWP:0xee␍␊
[06:33:09:925] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00␍␊
[06:33:09:931] mode:DIO, clock div:1␍␊
[06:33:09:933] load:0x3fff0030,len:1240␍␊
[06:33:09:936] load:0x40078000,len:13012␍␊
[06:33:09:936] load:0x40080400,len:3648␍␊
[06:33:09:939] entry 0x400805f8␍␊
```

So, going for the command line method, the automatic download mode does not seem to work and esptool times out waiting for the device. I then manually put the ESP32 into download mode by shorting IO0 (pin 25 per datasheet) to GND and cycling the on/off switch, and verified with esptool (git 65f86142 from today 20th-June-2022) and its chip_id command; That seems to be working:
```
user@host:~/path/esptool.git$ python3 esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset chip_id
esptool.py v4.2-dev
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ac:67:b2:1b:0d:18
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: ac:67:b2:1b:0d:18
Hard resetting via RTS pin...
```

I then ran:
```
user@host:~/path/esptool.git$ python3 esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 ../SparkFun_RTK_Firmware.git/Binaries/bin/RTK_Surveyor.ino.bootloader.bin 0x8000 ../SparkFun_RTK_Firmware.git/Binaries/bin/RTK_Surveyor.ino.partitions.bin 0xe000 ../SparkFun_RTK_Firmware.git/Binaries/bin/boot_app0.bin 0x10000 ../SparkFun_RTK_Firmware.git/Binaries/RTK_Surveyor_Firmware_v2_2.bin
esptool.py v4.2-dev
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ac:67:b2:1b:0d:18
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x001f1fff...
Compressed 17984 bytes to 12318...
Wrote 17984 bytes (12318 compressed) at 0x00001000 in 0.4 seconds (effective 360.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.0 seconds (effective 552.3 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 736.3 kbit/s)...
Hash of data verified.
Compressed 1971904 bytes to 1244381...
Wrote 1971904 bytes (1244381 compressed) at 0x00010000 in 21.3 seconds (effective 741.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Unfortunately the RTK Surveyor is now not working correctly the ESP32 never makes it to the firmware:
```
[06:57:14:437] rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)␍␊
[06:57:14:442] configsip: 0, SPIWP:0xee␍␊
[06:57:14:445] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00␍␊
[06:57:14:451] mode:DIO, clock div:1␍␊
[06:57:14:453] load:0x3fff0030,len:1240␍␊
[06:57:14:456] load:0x40078000,len:13012␍␊
[06:57:14:456] load:0x40080400,len:3648␍␊
[06:57:14:459] entry 0x400805f8␍␊
```

What went wrong? How to fix this?

LeeLeahy2, 2022-07-04:
There are two possible causes that I can easily think of:

1. Most likely you are using the wrong version of esptool.py, version 3.1.0 works for me
2. Alternatively you are using an old partition file: Binaries/bin/RTK_Surveyor.ino.partitions.bin

eshattow, 2022-07-05:
I tried again using esptool 3.1.0 with the documented virtual Ubuntu build environment:

```
$ ~/SparkFun/esptool-v3.1-linux-amd64/esptool --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 ~/SparkFun/RTK/Binaries/bin/RTK_Surveyor.ino.bootloader.bin 0x8000 ~/SparkFun/RTK/Binaries/bin/RTK_Surveyor.ino.partitions.bin 0xe000 ~/SparkFun/RTK/Binaries/bin/boot_app0.bin 0x10000 ~/SparkFun/RTK/Binaries/RTK_Surveyor_Firmware_v2_2.bin 
esptool.py v3.1
Serial port /dev/ttyUSB0
Connecting........_____....._____.
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ac:67:b2:1b:0d:18
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x001f1fff...
Compressed 17984 bytes to 12318...
Wrote 17984 bytes (12318 compressed) at 0x00001000 in 0.4 seconds (effective 356.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.0 seconds (effective 528.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 708.0 kbit/s)...
Hash of data verified.
Compressed 1971904 bytes to 1244381...
Wrote 1971904 bytes (1244381 compressed) at 0x00010000 in 21.2 seconds (effective 742.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

It is boot-looped still:

```
[17:26:43:903] rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)␍␊
[17:26:43:909] configsip: 0, SPIWP:0xee␍␊
[17:26:43:912] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00␍␊
[17:26:43:917] mode:DIO, clock div:1␍␊
[17:26:43:920] load:0x3fff0030,len:1240␍␊
[17:26:43:923] load:0x40078000,len:13012␍␊
[17:26:43:923] load:0x40080400,len:3648␍␊
[17:26:43:926] entry 0x400805f8␍␊
[17:26:43:932] ets Jun  8 2016 00:22:57␍␊
[17:26:43:932] ␍␊
[17:26:43:932] rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)␍␊
[17:26:43:937] configsip: 0, SPIWP:0xee␍␊
[17:26:43:940] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00␍␊
[17:26:43:945] mode:DIO, clock div:1␍␊
[17:26:43:948] load:0x3fff0030,len:1240␍␊
[17:26:43:951] load:0x40078000,len:13012␍␊
[17:26:43:951] load:0x40080400,len:3648␍␊
[17:26:43:954] entry 0x400805f8␍␊
[17:26:43:960] ets Jun  8 2016 00:22:57␍␊
```

Is auto-detected Flash size 4MB correct? Maybe this should be 16MB:
```
$ ~/SparkFun/esptool-v3.1-linux-amd64/esptool --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 16MB 0x1000 ~/SparkFun/RTK/Binaries/bin/RTK_Surveyor.ino.bootloader.bin 0x8000 ~/SparkFun/RTK/Binaries/bin/RTK_Surveyor.ino.partitions.bin 0xe000 ~/SparkFun/RTK/Binaries/bin/boot_app0.bin 0x10000 ~/SparkFun/RTK/Binaries/RTK_Surveyor_Firmware_v2_2.bin 
esptool.py v3.1
Serial port /dev/ttyUSB0
Connecting.......
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ac:67:b2:1b:0d:18
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x001f1fff...
Flash params set to 0x024f
Compressed 17984 bytes to 12318...
Wrote 17984 bytes (12318 compressed) at 0x00001000 in 0.4 seconds (effective 356.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.0 seconds (effective 527.7 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 707.4 kbit/s)...
Hash of data verified.
Compressed 1971904 bytes to 1244381...
Wrote 1971904 bytes (1244381 compressed) at 0x00010000 in 21.3 seconds (effective 742.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

So then it is boot looping but the message is different:

```
[17:32:07:235] E (613) spi_flash: Detected size(4096k) smaller than the size in the binary image header(16384k). Probe failed.␍␊
[17:32:07:243] ␍␊
[17:32:07:243] assert failed: do_core_init startup.c:298 (flash_ret == ESP_OK)␍␊
[17:32:07:248] ␍␊
[17:32:07:248] ␍␊
[17:32:07:248] Backtrace:0x40083da5:0x3ffe3aa00x40095309:0x3ffe3ac0 0x4009a965:0x3ffe3ae0 0x40119b1e:0x3ffe3c10 0x400834be:0x3ffe3c40 0x40078f0e:0x3ffe3c90  |<-CORRUPTED␍␊
[17:32:07:262] ␍␊
[17:32:07:262] ␍␊
[17:32:07:262] ␍␊
[17:32:07:262] ␍␊
[17:32:07:262] ELF file SHA256: 0000000000000000␍␊
[17:32:07:265] ␍␊
[17:32:07:268] Rebooting...␍␊
[17:32:07:268] ets Jun  8 2016 00:22:57␍␊
[17:32:07:271] ␍␊
[17:32:07:271] rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)␍␊
[17:32:07:277] configsip: 0, SPIWP:0xee␍␊
[17:32:07:277] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00␍␊
[17:32:07:285] mode:DIO, clock div:1␍␊
[17:32:07:285] load:0x3fff0030,len:1240␍␊
[17:32:07:288] load:0x40078000,len:13012␍␊
[17:32:07:290] load:0x40080400,len:3648␍␊
[17:32:07:295] entry 0x400805f8␍␊
```

is Binaries/bin/RTK_Surveyor.ino.partitions.bin the wrong file? Did the hardware fail somehow?

nseidle, 2022-07-05:
Hi @eshattow - Thanks for reporting. 

You are correct: you're unable to update the firmware because you have a 4MB module. The good news is that, with a bit of a wait, you'll be able to enjoy all the new features we're adding. The bad news is that you'll need to update your firmware using the serial CLI or GUI methods going forward.

Longer explanation:

We initially sold the RTK Surveyor with the 4MB module. We then changed the BOM to use the 16MB module. Firmware v2.3-Dev is sitting around 2.1MB. Because an ESP32 needs two equal partitions to support OTA for WiFi and SD updates, 4MB is not large enough to support the firmware as is.

Currently, we are compiling the RTK firmware for the 16MB target. Because you are the first to report a 4MB upgrade issue in the field, we haven't been compiling a separate binary for 4MB users. We'll start compiling a 2nd, 4MB compatible binary for you and other 4MB users shortly.

eshattow, 2022-07-05:
That's good news, it is not difficult to update the firmware with esptool. However, I don't know how to get the esp32 into download mode from Linux automatically. I've had to manually bridge pin IO0 low to GND (in as much as I've been able to that does mean taking the case off, disassembly, some skilled work to probe the correct pads on the esp32 module). The schematic says this is tied to RTS on the usb serial module. I think as esptool would be required for future updates the [hookup guide](https://learn.sparkfun.com/tutorials/sparkfun-rtk-surveyor-hookup-guide/all) perhaps should point out visually which connection on the assembly should be held low manually if the automatic method of putting esp32 into download mode is not working. It would also be worth figuring out why here that automatic method is not working, but that is another issue and I guess may be dependent on OS drivers. The other thing to document would be an upgrade procedure of the 4MB esp32 module to the newer 16MB esp32 ; is it a straightforward module swap and the part is available in single quantity? If the automatic selection of download mode is not totally reliable due to design or outside factors (host OS drivers?) and the case must come off (and board unfastened for access to underside) to upgrade, then the correct procedure for manually inducing download mode and also a procedure to swap the esp32 module for an OTA-supporting part would be good to include with the release of these new 4MB-supporting firmwares.

Thank you very much for supporting the older versions of the product :-)

nseidle, 2022-07-05:
> However, I don't know how to get the esp32 into download mode from Linux automatically.

This is odd. Perhaps @LeeLeahy2 can give you some guidance and I'm currently windows based.

> upgrade procedure of the 4MB esp32 module to the newer 16MB esp32 ; is it a straightforward module swap and the part is available in single quantity

The 16MB WROOM is [readily available](https://www.sparkfun.com/products/17830) but without some pretty sophisticated tools, and a lot of experience, it's not trivial to pull a module from a 4-layer PCB. For now, I would say there is no upgrade path. But once we get your auto-boot situation fixed, upgrading is really not needed.

> Thank you very much for supporting the older versions of the product :-)

Absolutely! We'll do our best.

nseidle, 2022-07-08:
Good news. @PaulZC has updated the GUI to include partitions for both the 16MB and 4MB devices. The GUI also autodetects which module is seen and selects the appropriate partition file. Looks like you're running straight CLI so, FYI there is a flash_id command (https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/basic-commands.html#read-spi-flash-id-flash-id) to check the device flash size. We know it's 4MB.

Next, you need to direct esptool to use the correct partition file. The 4MB file is here (https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/release_candidate/Uploader_GUI/RTK_Surveyor_Partitions_4MB.bin) for now. We'll get it merged into main once we do a release.

The command structure for you would be something like:

    python3 esptool.py --chip esp32 --port COM6 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 RTK_Surveyor.ino.bootloader.bin 0x8000 RTK_Surveyor_Partitions_4MB.bin 0xe000 boot_app0.bin 0x10000 RTK_Surveyor_Firmware_vxx.bin

Here is the release candidate from today: [RTK_Surveyor_Firmware_v2_3RC-July8.zip](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/9073040/RTK_Surveyor_Firmware_v2_3RC-July8.zip)

I believe we can have one binary with different partition bins and allow 4MB users to continue to update (without OTA support). I've tested this but please confirm that you can load the RC binary and have your Surveyor correctly begin running. This RC is not completely ready for release (there may be some bugs) but you shouldn't have the rolling reboot issue.

eshattow, 2022-07-08:
`$ git checkout release_candidate && git pull d9bf85c0
`
Note that IO0 is manually pulled low to GND and trigger esp32 download mode during the run of esptool command:
```
$ python3 ./Uploader_GUI/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 ./Uploader_GUI/RTK_Surveyor.ino.bootloader.bin 0x8000 ./Uploader_GUI/RTK_Surveyor_Partitions_4MB.bin 0xe000 ./Uploader_GUI/boot_app0.bin 0x10000 /tmp/RTK_Surveyor_Firmware_v2_3RC-July8.bin
esptool.py v3.3
Serial port /dev/ttyUSB0
Connecting.............................
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ac:67:b2:1b:0d:18
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00202fff...
Compressed 17984 bytes to 12318...
Wrote 17984 bytes (12318 compressed) at 0x00001000 in 0.4 seconds (effective 363.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 119...
Wrote 3072 bytes (119 compressed) at 0x00008000 in 0.0 seconds (effective 505.8 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 707.4 kbit/s)...
Hash of data verified.
Compressed 2042544 bytes to 1275432...
Wrote 2042544 bytes (1275432 compressed) at 0x00010000 in 22.0 seconds (effective 741.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
The firmware runs successfully "SparkFun RTK Surveyor v2.3-Jul  8 2022".

nseidle, 2022-07-08:
> The firmware runs successfully

Excellent! I'm not sure why your CH340 drivers are not allowing the correct pulsing of RTS and DTR to get the device into boot mode. Please have a look at your [CH340 driver tutorial](https://learn.sparkfun.com/tutorials/how-to-install-ch340-drivers/all#linux), it may help.

Closing since the primary issue has been fixed. Please let us know if other issues persist.

closed 2022-07-08 by nseidle