sparkfun/SparkFun_RTK_Firmware #244: no rtk after update firmware

montoresjr, 2022-07-22:
### Subject of the issue
Hello, I'm new to the forum, I apologize in advance for my mistakes in English :) but, I need help with my receivers. I have 2 RTK Surveyors that don't get rtk solution after upgrading to hpg 1.32, tried downgrade unsuccessfully, replace cables and antennas with new ones but ubx shows too much noise. I attach the configuration files and the raw data of both receivers.

### Your workbench
BASE:
RTK FIRMWARE VERSION= V1.12 - MAY 3 2022
ROVER:
RTK FIRMWARE VERSION= V1.12 - MAY 3 2022
BASE:
ZED FIRMWARE VERSION= HPG 1.32
ROVER:
ZED FIRMWARE VERSION= HPG 1.12
BASE FILES:
https://drive.google.com/drive/folders/1Vsln60eKUv5esQLTz6g_tmhFlukImhy7?usp=sharing
ROVER FILES:
https://drive.google.com/drive/folders/1FztuU3j_sW10iQQ0gFiuU0cT2E-GKBR9?usp=sharing

### Steps to reproduce
I CONNECT RECEIVERS, CABLES, ANTENNAS AND RADIOS.
TURN ON BASE
I WAIT FOR THE RED LED (BASE) TO STAY STEADY
STARTING THE ROVER

### Expected behavior
CONNECT RECEIVERS
WAIT FOR THE BASE TO FINISH THE SURVEY
HAVE THE BASE ISSUE RTCM CORRECTIONS
THAT THE ROVER RECEIVE RTCM AND ACHIEVE RTK SOLUTION
KEEP THE RTK LED STEADY
LET THE 1CM LED REMAIN FIXED

### Actual behavior
CONNECT RECEIVERS
BASE ENDS SURVEY AND KEEPS RED LED (BASE) STEADY.
STARTING THE ROOVER.
SPOKES ARE PAIRED
RTK LED (WHITE) IS BLINKING
NO FIXED RTK, ONLY DGPS AND OCCASIONALLY FLOATING RTK

nseidle, 2022-08-08:
Please consider using our Windows GUI to update the ZED F9x firmware. Please see: https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update/#updating-u-blox-firmware

Using the GUI I am able to upgrade and downgrade the ZED successfully. However, I don't believe the ZED firmware is the issue.

> OCCASIONALLY FLOATING RTK

This would indicate the two systems are working. Perhaps the radios are dropping some of the RTCM data, it's hard to say without more information. I recommend you try a different radio for testing (use WiFi or cellular). If I were to troubleshoot the system I would use u-center to inspect the incoming RTCM packets on the Rover to verify they were coming through correctly:

[image]

tonycanike, 2022-08-09:
@montoresjr , which radios are you using?  I use the Holybros that Sparkfun sells.  https://www.sparkfun.com/products/19032

When I have problems getting an RTK fix, I first check that the radios are communicating.  Sometimes the cables come loose.
The green LED on each radio should be solid, and the red LED on each radio should flash as the data flows from the base to the rover.

montoresjr, 2022-08-11:
Hello guys!!! Thanks for the comments and for the support. Yesterday afternoon while I was disassembling the antenna from the base, I heard something strange inside the antenna, I disarmed it and what a surprise! it was full of rust, I don't understand how it could happen since it hasn't gotten wet but I want to think that this is the cause of so much noise in the raw data files. For now, I already ordered 2 new antennas, as soon as they arrive I will do the tests again and I will tell you how it went ;)

nseidle, 2022-08-12:
Ah! That would make sense. I'm going to close this issue since it's not firmware related. If you run into new issues please let us know.

closed 2022-08-12 by nseidle