sparkfun/SparkFun_RTK_Firmware #245: Add SEN-19921 to RTK Facet, get new NMEA messages with heading data

XPav, 2022-07-27:
The QWIIC port on the RTK Facet is unused, but I believe the new SEN-19921 Magnetometer is a great thing to add to the system.

This would allow the generation of new NMEA message that need heading, such as:

- HDG
- HDT
- THS
- VTG? 

Then it's just a "simple" hop-skip-jump to add an accelerometer and make a full-up INS!

nseidle, 2022-07-30:
The RTK Facet already has a LIS2DH12 inside: https://www.sparkfun.com/products/15760

We are considering adding the MMC5983MA magnetometer. While the hardware add is trivial, the math and NMEA modifications to get to a tilt compensated lat/long are not. It's been a fun challenge actually! 

We have no plans on implementing an inertial nav system. Get the ZED-F9R in the RTK Express Plus if you need that.

For all readers:

* Are you more interested in NMEA heading sentences (HDG, HDT, etc)?
* Are you more interested in automatically corrected lat/long for surveying a spot next to a building or other monument where you cannot get the unit vertical?

I suspect if we're going to have to modify the NMEA sentences coming out of the ZED, it's fairly trivial to add additional NMEA sentences for heading. I'm just a bit hesitant to do so.

gdt, 2022-07-30:
(I am most interested in being to able to use Free Software data collectors.)

I'm not super interested in heading, because I don't believe it will be very accurate, due to the nature of magnetic fields and local anomalies.

Being able to output the position of the tip of the prism pole would be huge.  I would do that by reading UBX messages, converting the vector to XYZ, adjusting, and then back to LLh (and maybe LLH).  But it may be that doing that offset from LLH in the first place is just as easy - it's just math and parsing.

This would require entering a "HI", dealing with L1/L2 phase center vertical offsets, and also some sort of accelerometer cal, maybe 4 azimuths while leveling the prism pole (with a bubble level that has been adjusted) and pushing  a cal entry in web UI.  Extra tricky for non-facet in a mount, but would be nice to support that too.

tonycanike, 2022-07-30:
Nathan, thanks for asking.

I am also more interested in the getting the position of the tip of the 
pole when the pole is not vertical.  Besides being able to get a reading 
when you can't get the pole vertical, if someone (as I do at times) need 
to take 100's of shots in an afternoon for a topographic survey, not 
having to level..err plumb... the pole for every shot would speed things up.

How do the expensive land survey GNSS receivers do this and how the data 
is reported to the data collector?  With a vertical pole, it's my 
understanding the GNSS receiver reports the coordinates (via NMEA) of 
its antenna phase center and the data collector does the little bit of 
math to account for the HI.

For non-level poles,

 1. Does the GNSS receiver report the tilt & phase center coordinates,
    and then does the data collector take that info, along with the HI
    entered into the DC, and do all the math to get the coords for the
    tip of the pole?
 2. Or does the data collector tell the GNSS receiver the HI,  the GNSS
    receiver does the hard math math, and then the receiver reports the
    coordinates of the tip to the DC?    But the DC is used to getting
    the phase center coordinates and not the pole tip coordinates...so
    what happens?

More simply, I'm asking if the data collectors, and not the GNSS 
receivers, do the hard math of the tilt.   And if so, does the Sparkfun 
RTK firmware really need to do the hard math, or will the (expensive 
commercial) data collectors do it?

I use commercial data collector SW (SurvPC) and I'd be happy to make use 
of all the features that I paid for!

---

I'm not that interested in getting heading; I don't do work that 
requires heading, etc.

nseidle, 2022-07-30:
Thanks gdt and Tony!

> Extra tricky for non-facet in a mount, but would be nice to support that too.

Agreed. Tilt compensation would be limited to known static reference frames, ie: Facet and Facet L-Band. Surveyor/Express support with any kind of accuracy is next to impossible.

> More simply, I'm asking if the data collectors, and not the GNSS receivers, do the hard math of the tilt.   And if so, does the Sparkfun RTK firmware really need to do the hard math, or will the (expensive commercial) data collectors do it?

Hmm, I never thought about this. I assumed the data collector software was dumb/merely showing NMEA coordinates, but perhaps I should not. Tony, please dig and let me know if you have any inkling. Perhaps all we need to do is provide the additional NMEA sentences via Bluetooth.

We will likely mount a mag + accelerometer to the center of the retention plate with a small Qwiic cable to connect from the plate to the main board (we added a Qwiic connector to the main board just for future purposes such as this one). This will locate the frame of reference center to the prism pole and simplify our math. We could try to use the built-in accel on the display board but because it's ~40mm off center, it's not ideal.

I am happy to provide hardware to whoever wants to take a run at the firmware. It's on our roadmap but low at the moment.

tonycanike, 2022-07-30:
Just a preliminary reply - looking at the SurvPC manual, section 5.10, pp 193-198,  if the GNSS has tilt sensors that SurvPC supports, SurvPC will display a virtual bubble to assist in levelling the pole.  The manual states that if GNSS receiver is not level, "the software"  (I presume they mean SurvPC) will correct for the non-level poles.  There are details about not being able to correct due to magnetic interference, motion of the pole, etc.  I'll dig into this more when I have more time.

gdt, 2022-08-09:
About the question about how professional data collectors do things: I think that's a more complicated question, but well worth digging into.  Are they really getting standards-compliant NMEA out of some surveying-grade RTK rover?  I would sort of expect that in e.g. a Trimble system that the GNSS unit and data collector would be speaking some Trimble proprietary protocol, and that you can't just plug a Leica data collector into a Trimble receiver.

nseidle, 2023-01-19:
We will continue to explore how to implement a tilt-compensated function into future versions of the RTK product line. For now, I'm closing this issue because it's far more complicated than simply adding an external mag.

closed 2023-01-19 by nseidle