sparkfun/SparkFun_RTK_Firmware #257: sparkfun rtk express

skleins000, 2022-08-01:
Dropping off wifi.

I am using a sparkfun express survey unit as a permanent base to connect to RTK2go.com Port 2101 for RTK correction data, however the base randomly just drops off the Wifi Connection and goes from Casting to Xmitting on the LCD screen on the device. The wifi it is on doesn't require a password so i am running version RTK_Surveyor_Firmware_v1_10-RC-Feb16.

Let me know if there is anything i can do. I talked with the Internet guys on the ground and they couldn't find anything blocking the path of its ability to access wifi and the website. This is what makes me think there is something wrong with the base as i think it should connect permanently. Whenever it drops off, i go back up and restart it and then it reconnect and starts casting, then randomly it drops off. Has been as short as 20 min and as long as 12hours.

gdt, 2022-08-01:
I'm not from sparkfun and am unclear on doctrine, but I would suggest updating to the most recent release.   When I maintain things I never want to debug old versions.

skleins000, 2022-08-01:
Yes I know that but I had to use that version because the Wi-Fi hotspot I'm using doesn't require a password and if I didn't enter a password in the slot it wouldn't connect with the most up-to-date version. That's why I'm using the older version but now I have the problem of it dropping off Wi-Fi randomly.

Sent from Nine<http://www.9folders.com/>

nseidle, 2022-08-01:
@skleins000 - Understood. I too have seen issues with an RTK Surveyor dropping WiFi connection over a few hours but haven't had the chance to focus on the problem. We're working to get v2.3 out shortly. We'll work on this issue shortly.

skleins000, 2022-08-09:
Just wondering when i can expect to get this issue fixed as i was hopping to use this setup by the middle of August. Is that realistic for you because otherwise i could return for refund from sparkfun and i might by the Emlid base station.

nseidle, 2022-08-10:
Please give [v2.4 release candidate](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.4RC) a try.

skleins000, 2022-08-11:
ok, i am having a hard time getting the release to upload onto the base. Do you have any tips, i am not able to get it to connect using the com port although it shows up as a com device?

nseidle, 2022-08-13:
Update instructions are [here](https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update/). I recommend the [GUI](https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update/#updating-firmware-using-windows-gui) if you're on Windows.

If you're still having problems, please let me know more specifics of what you're seeing and having problems with.

nseidle, 2022-08-18:
I believe this has been fixed with v2.4 RC Aug 18th or later. Please open a new issue if you're still seeing problems.

closed 2022-08-18 by nseidle