sparkfun/SparkFun_RTK_Firmware #266: RTK Surveyor 4MB esptool Linux does not enter bootloader mode

eshattow, 2022-08-04:
Schematic says that RTS should bring IO0 low for bootloader mode and DTR low for reset.

When using esptool the target RTK Surveyor 4MB does not enter bootloader mode.

Using a probe to bring IO0 low to GND does enable bootloader mode. Is this a Linux driver problem not activating RTS?

Debian GNU/Linux bookworm/sid
Linux zontar 5.18.0-2-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.18.5-1 (2022-06-16) x86_64 GNU/Linux
$ lsmod | grep usbserial
usbserial              61440  3 ch341
usbcore               327680  8 usbserial,xhci_hcd,usbhid,usb_storage,btusb,xhci_pci,uas,ch341

eshattow, 2022-08-04:
> Please have a look at your [CH340 driver tutorial](https://learn.sparkfun.com/tutorials/how-to-install-ch340-drivers/all#linux), it may help.
_Originally posted by @nseidle in https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/207#issuecomment-1179278672_

I've tried that https://github.com/juliagoda/CH341SER.git driver, also the https://github.com/WCHSoftGroup/ch341ser_linux.git (with some minor fixes for compilation), and none are moving the device to bootloader mode.

Next troubleshooting step I suppose is to connect with another OS to determine if it works at all on the same hardware.

eshattow, 2022-08-04:
Attempted with a fresh KVM virt-manager install of Windows 10 LTSC. Installed firefox, get snapshot of SparkFun_RTK_Firmware git repo, install Python for windows, pip3 install pyqt5 pyserial, install CH341SER driver from wch-ic.com vendor. From host OS add blacklist lines to /etc/modprobe.d/chserial-blacklist.conf for usbserial and ch340 ch341 ch34x just to be thorough. Connect USB-C cable to RTK Surveyor and turn power switch on, then in virt-manager set USB redirector for the device. Guest OS Windows 10 makes its hardware detected noise and verify in the hardware manager. Run from SparkFun_RTK_Firmware: python RTK_Firmware_Uploader_GUI.py and select the 2.3 firmware binary, ch341 COM device, and "Upload Firmware" button.

Hardware looks like it resets.

Output follows:
```
Ports Refreshed

Detecting flash size

esptool.py v3.3
Serial port \\.\COM3
Connecting...
Device PID identification is only supported on COM and /dev/ serial ports.
...................................

Failed to connect to ESP32: Wrong boot mode detected (0x13)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```
So is this a hardware problem with the Sparkfun RTK Surveyor 4MB?

nseidle, 2022-08-04:
Perhaps @LeeLeahy2 can add some insight as he's primarily on Linux.

I believe this is a software/driver issue with the CH340 and Linux. The RTS and DTR must be driven with specific wave forms to set BOOT and EN accurately on the ESP32. esptool.py controls RTS and DTR in the correct ways, but it is up to the driver on any given platform to correctly follow them. I believe we can eliminate the hardware and the CH340 IC itself as these are working on other platforms.

Here are some additional things to try (in case you haven't already): https://learn.sparkfun.com/tutorials/how-to-install-ch340-drivers/all#linux

LeeLeahy2, 2022-08-04:
The procedure that I have been using to build the firmware is documented [here](https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update/#ubuntu-2004).  We have been using the 2.0.2 version of the ESP32 Core which includes version 3.1.0 of the esptool.py.  I have not needed to include any special drivers beyond what is included with Ubuntu 20.04.  This version of esptool.py seems to require python3 to execute successfully as documented in the new-firmware.sh shell script.

Nathan, does he also need a different partition file?  If so, where in the release tree is the 4MB partition file?  If this file is needed then a change is necessary to the new-firmware.sh shell script to replace RTK_Surveyor.ino.partitions.bin with the 4MB partition file.

nseidle, 2022-08-04:
Note the OP is having problems just getting the ESP32 into boot mode. I don't believe this is partition file related, but @eshattow will need it eventually.

Shoot, the docs should have had a link but they don't. Updated [now](https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update/#updating-4mb-surveyors)

4MB partition file is [here](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/main/Binaries/bin/RTK_Surveyor_Partitions_4MB.bin).

nseidle, 2022-08-08:
Hi @eshattow - Let us know if you were able to properly control the CH340 via Linux.

eshattow, 2022-08-09:
> Hi @eshattow - Let us know if you were able to properly control the CH340 via Linux.

No, the reset is happening but the bootloader mode is not activated. I also tried this from Windows 10 in a virtual machine (on the same Linux host as tested) and USB passthrough so the Windows Driver was in control, and there was no change from behavior of Linux 5.18.0

Should I try to find some different computer to test against? Perhaps Intel vs AMD USB controller, or, I don't know what else to try?

How can I verify if this is a driver problem or a RTK Surveyor 4MB problem?

nseidle, 2022-08-10:
Harumph. I'm not sure what to recommend at this point. 

> How can I verify if this is a driver problem or a RTK Surveyor 4MB problem?

It sounds like you've tested really everything. I'm not sure how to test without probing individual pins with a logic analyzer with esptool starts up.

 I'm happy to do a return and have a look at the unit for you to confirm there's nothing wrong with the hardware. Let me know. This is a head scratcher.

eshattow, 2022-08-16:
@nseidle Sure let's do that. I've submitted a request at https://www.sparkfun.com/returns

Funda, 2022-08-17:
We have your return ticket open @eshattow. I will get you taken care of momentarily.

LeeLeahy2, 2022-08-18:
What version of Linux are you running?

I did an install on an old version of Ubuntu and I found that I needed to reduce the baud rate to get the boot loader to work successfully.  I set the baud rate to 230400 and was able to successfully upload the image to the RTK Express.  The baud rate is set in the Arduino IDE using the Tools/Upload Speed menu item.  The baud rate is also hard coded in the new-firmware.sh shell script file.

The latest release candidate contains the proper partition .bin file for the 4MB unit.  Please edit the new-firmware.sh shell script file when programming a 4MB RTK Express to use the 4MB version of the partition file.

nseidle, 2022-10-10:
Hi @eshattow - I've got your unit and it bootloads via software as expected. The RTK firmware GUI is able to load new code and I see the unit resetting via the toggling of the port via TeraTerm. This all indicates to me that the hardware is working as designed and expected. At this point I have to chalk this up to a CH340 driver / Linux problem. We have other Linux users able to load code but I am not an expert so I'm afraid my support is limited. 

WRT hardware: I noticed two things about your unit:

* The base switch wires have been either cut or broken. Would you like us to repair them?
* This RTK Surveyor PCB version uses the original SMD version of the antenna connector which can rip off if the unit is dropped. In general, we replace this with a much more robust bulkhead connector. Would you like us to modify your unit to prevent future antenna connector failure?
[image]

eshattow, 2022-10-11:
Hi Nathan, that is strange about the base switch wires, not something
intentional by me. Trivial to repair (I agree).
1. While you have the device can you verify updating firmware works with a
host PC running some bootable Linux live image ? I will try the same Linux
live image here, for example:
https://cdimage.debian.org/cdimage/unofficial/non-free/cd-including-firmware/11.5.0-live+nonfree/amd64/iso-hybrid/debian-live-11.5.0-amd64-gnome+nonfree.iso
Boot PC w/ Ventoy USB loader to GNU/Linux Live
Complete welcome wizard connecting wifi for Internet
Gnome Terminal:
sudo apt update
sudo apt install git python3-serial python3-pyqt5.qtserialport
git clone https://github.com/sparkfun/SparkFun_RTK_Firmware_Binaries.git
sudo python3
./SparkFun_RTK_Firmware_Binaries/Uploader_GUI/RTK_Firmware_Uploader_GUI.py

2. While updating for a bulkhead antenna connector can the ESP module be
exchanged with the current production module that can update from SD card?
I understand if there is a cost for this, the functionality was lost as the
firmware improved and has grown in size, that's how things go...

Although I'm glad to hear that the firmware update is working as expected
for you, I am hoping to uncover the cause of why it does not work for me.

Thank you!  -E Shattow

nseidle, 2022-10-11:
> While you have the device can you verify updating firmware works with a host PC running some bootable Linux live image ?

I attempted to boot to a bootable USB drive with the image you point to but hit a GRUB failure and need to return to other work. I'm sorry but I'm not going to be much help troubleshooting the driver issue within Linux. I maintain the CH340 needs appropriate drivers so that the DTR and RTS lines are driven via software in order to reset and put the ESP32 into boot mode. It sounds like you have a workable (albeit tedious) hardware solution for the time being.

> While updating for a bulkhead antenna connector can the ESP module be exchanged with the current production module that can update from SD card?

A good question but unfortunately it is next to impossible to hot-air remove the ESP32 from the board without damaging the PCB or other parts of the device.

I'll have it repaired and boxed up for return.

eshattow, 2022-10-11:
Nathan, I appreciate your time, however is booting Linux and at least
trying this too much to ask? That is a standard release ISO of Debian
(albeit a variety with non-free drivers added - I have a wifi chipset here
that requires this) and is the same as what I have tested locally.

Please, if this is not done, we have learned nothing about the circumstance
of the failure.

E Shattow

nseidle, 2022-10-12:
Hi Eshattow, for a luddite like myself, bringing up Debian from a bootable USB stick under our corporate environment is beyond what I'm capable of. We have instructions for setting up [Ubuntu](https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update/#ubuntu-2004) that is capable of software controlled bootloading. I hope these instructions can be of some use. If there is additional documentation that can be added to help other users, I welcome adding it.

closed 2022-10-12 by nseidle

eshattow, 2022-10-12:
Yes, also by that description it sounds as though it was only verified that
the unit resets, which is true it resets for me as well. Were you able to
confirm loading a different version of firmware and verify that upload  was
successful through a connection to the device? It did also "appear" to
function for me via the GUI tools but was not successful. I am not
convinced this ticket merits closing until the device is successfully
tested with Linux (Ubuntu or otherwise) as that is what the ticket is about.

nseidle, 2022-10-12:
Yes, unit firmware was loaded with v2.3 to match the previous firmware that was on the unit.

eshattow, 2023-02-13:
Confirming #361 resolves this trouble. Thanks again!