sparkfun/SparkFun_RTK_Firmware #268: Exiting Tera Term causes Facet to shutdown

tonycanike, 2022-08-05:
### Subject of the issue
After a serial configuration of a Facet in Rover mode, exiting Tera Term causes the rover to shutdown.

### Your workbench
RTK Firmware 2.3
No Bluetooth connection, no NTRIP enabled
Powered via charged internal battery and USB cable to PC
Windows 10
Tera Term 4.106

### Steps to reproduce
1. Power on Facet, ensure in Rover mode
2, Connect USB Cable to PC.  Windows makes hardware connected noise.
3. Launch Tera Term, connect to CH340 COM port, set baud to 115200.  See the usual messages.
4, press key, enter serial config menu.  Facet display shows Serial Config
5, perform configurations (e.g. set PPP logging defaults)
6. x all the way out of the serial menu
7. The usual Accuracy/Battery/UBX File size messages begin appearing in Tera Term. UBX file size increasing at a reasonable rate.
8. Facet display exits Serial Config and shows normal display, including logging SD Card icon.
9. Exit Tera Term.
10. Sparkfun logo appears briefly on Facet.
11. Facet shuts down.
12. Windows makes hardware disconnected noise.

### Expected behavior
The Facet should not shut down when Tera Term exits.

### Actual behavior
The Facet shuts down when Tera Term is closed.

NOTES:  
a.  If I disconnect the USB cable first, and then exit Tera Term, the Facet does NOT shut down.  This is my work-around.
b.  Putty doesn't power down the Facet when exiting.
c.  The Arduino serial monitor doesn't power down the Facet when existing.

nseidle, 2022-08-08:
Hi Tony - I've seen this to. I have been attributing it to how TeraTerminal handles RTS/DTR when it opens/closes a terminal. Toggling these lines are what reset and put the ESP32 into bootloader mode. This is not RTK firmware related, it's more hardware related and depends on Tera Terminal which we can't control. 

Long term you'll decide if you want to use TeraTerminal as your terminal. I think it's great for long term logs and has come in handy a few times for force resetting a unit.

closed 2022-08-08 by nseidle

tonycanike, 2022-08-08:
I figured it might be something like that.  Thanks for the response!

tonycanike, 2022-08-09:
I updated configure_with_serial.md to document this behavior.