sparkfun/SparkFun_RTK_Firmware #272: RC v2.4 is available

nseidle, 2022-08-08:
FYI - RC 2.4 is over [here](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.4RC)

Currently, it's just a few bug fixes but this is the first RC that is produced from our Github actions + Arduino CLI so it's not 'just Nate's computer of the day'. Please feel free to give it a test. There are plenty of known issues within it so please don't run it before going into the field.

I really wanted to report that ESP Now was working, and it's *so* close, but there are some really bad interactions when attempting to use Bluetooth and WiFi and the ESP Now 2.4GHz radio at the same time. TBD.

nseidle, 2022-08-10:
ESP-Now now working. This enables point to point and point to multipoint RTCM broadcast between RTK units without the need for an external radio, WiFi, cellular, or anything else. The downside is that the range is limited to what you would expect between your laptop and AP.

All three radios can be on simultaneously: WiFi and BT and ESP Now. It's pretty amazing this ESP32.

Icons in the top left corner have been re-arranged a bit. We've added up/down arrow indicators for each icon (WiFi, BT, ESP Now) depending on if RTCM is coming in (down-load) or begin sent out (up-load).

[image]

On the right is a unit setup as fixed base, pushing RTCM to WiFi, pushing data to ESP Now, and waiting for incoming BT connections.
On the left is a unit setup as rover, receiving RTCM over ESP Now, with RTK Fix, logging Custom messages, waiting for incoming Bluetooth.

Nightglobe, 2022-08-13:
I managed to put the firmware on rtk facet and finally I managed to get the Ntrip client to log in to the service provider on wifi and to read the corrections on bluetooth through my measuring program, Surpad and Surveymaster if you still implement and transmit the corrections to nmea at a tcp port I can say that you have reached and even surpassed Emlid Rs2. Bravo, I have 4 facets taken from you, two from Europe and two from Mouser, but the European branch

nseidle, 2022-08-13:
Thank you @Nightglobe! We still have many bugs to fix so please use v2,4 with caution in the field until we do a full release. It is good to hear you are interested in NMEA over TCP as well. We will get it on our roadmap.

gdt, 2022-08-14:
NMEA over TCP is #96.

I think that this release, with NTRIP over WiFi and NMEA over Bluetooth SPP, should work with QField.  If so, that's very excitign news as I think it's the first time it works with a Free Software android client.  I will be attempting to test soon.

gdt, 2022-08-17:
WIth the August 15th binary, I have succeeded in configuring NTRIP over wifi to MaCORS, and at the same time connecting QField via Bluetooth SPP, and seeing it display Float RTK.   The positions look reasonable.   This is inside, but that proves that the F9P is getting reference data.  I will try outside, but this is a huge success!  Thank you for making wifi/ntrip and bluetooth-spp work at the same time -- there is now a way to use this with an open source Android program.   (I am having some minor issues, but I'll ticket those later.)

nseidle, 2022-08-27:
v2.4 is released

closed 2022-08-27 by nseidle