sparkfun/SparkFun_RTK_Firmware #275: Firmware update from CLI failed

pyrog, 2022-08-11:
Updating firmware from CLI failed following the [Updating Firmware From CLI : macOS / Linux]( https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update) doc section.

```
python3 esptool.py --chip esp32 --port /dev/tty.usbserial-1410 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 ./bin/RTK_Surveyor.ino.bootloader.bin 0x8000 ./bin/RTK_Surveyor_Partitions_16MB.bin 0xe000 ./bin/boot_app0.bin 0x10000 ./bin/RTK_Surveyor_Firmware_RC-Aug_11_2022.bin 
esptool.py v3.3
Serial port /dev/tty.usbserial-1410
Connecting....
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a8:03:2a:f3:e5:34
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Traceback (most recent call last):
  File "esptool.py", line 5387, in <module>
    _main()
  File "esptool.py", line 5380, in _main
    main()
  File "esptool.py", line 4800, in main
    detect_flash_size(esp, args)
  File "esptool.py", line 3816, in detect_flash_size
    flash_id = esp.flash_id()
  File "esptool.py", line 835, in flash_id
    return self.run_spiflash_command(SPIFLASH_RDID, b"", 24)
  File "esptool.py", line 1131, in run_spiflash_command
    old_spi_usr = self.read_reg(SPI_USR_REG)
  File "esptool.py", line 708, in read_reg
    val, data = self.command(self.ESP_READ_REG, struct.pack('<I', addr), timeout=timeout)
  File "esptool.py", line 468, in command
    p = self.read()
  File "esptool.py", line 413, in read
    return next(self._slip_reader)
StopIteration

```

### Your workbench
* Command ran under macOS
* Tested on RTK Express (16 Mb) and RTK Surveyor (4 Mb)

### How to solved this issue
Simply set baudrate to `115200` according to the following screen copy😃 

[image]

nseidle, 2022-08-12:
Hmm, it's good to note that 921600bps may not work on Mac. The screen shot is from ublox update tool. esptool should work at a variety of baud rates.

gdt, 2022-08-17:
I just did a successful update at 115200, after failing to use the OTA/wifi method (1.12 -> 2.3), via the commandline esptool -- which was easier than needing to build qt5 which is massive.  A subsequent OTA hung in the browser but succeeded and the device rebooted and didn't enter wifi config, so it was hard to tell.

According to kermit the highest available is 230400.  I have not tried that.

I would suggest changing the instructions to suggest 115200.  It doesn't take that long, and it's the highest normal baud rate from the old days, before the more recent crazy-fast speeds, and I'm a bit suspicious of reliability of megabit over serial, even if it is USB/serial and short.   There are enough woes in updating that I think it's better to wait 3 minutes and win than to have to cycle back for more thought.

nseidle, 2022-08-18:
Docs [updated](https://github.com/sparkfun/SparkFun_RTK_Firmware/commit/20c6f516e0439e8fa8e0db0184e1c2f75801358f).

closed 2022-08-18 by nseidle