sparkfun/SparkFun_RTK_Firmware #282: Can't compile from source

XPav, 2022-08-16:
Following the instructions, I encountered the following problems building from source (tried both 2.3 main and 2.4 RC)

1. PubSubClient is not built-in despite the comment, required installation, no link in the source
2. ArduinoJson.h is not called out in the instructions, but the link was in the source

But those are easy to fix.

When I go to build, I get:
```

WARNING: library JC_Button claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
In file included from C:\Users\AlexPavloff\Documents\Arduino\SparkFun_RTK_Firmware\Firmware\RTK_Surveyor\RTK_Surveyor.ino:112:
C:\Users\AlexPavloff\Documents\Arduino\libraries\SdFat\src/SdFat.h:438:2: warning: #warning File not defined because __has__include(FS.h) [-Wcpp]
 #warning File not defined because __has__include(FS.h)
  ^~~~~~~
text section exceeds available space in boardSketch uses 1990809 bytes (101%) of program storage space. Maximum is 1966080 bytes.

Global variables use 80240 bytes (24%) of dynamic memory, leaving 247440 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
Error compiling for board ESP32 Dev Module.
```
For now, not needing Bluetooth, I commented out ``COMPILE_BT`` and went on.

nseidle, 2022-08-18:
Good catch. [Updated](https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update/#windows_1).

The sketch size is because you didn't overwrite the partition file. See step 3 'Change Partition Table'.

XPav, 2022-08-19:
I did Step 3, and just checked it again.  

[image]

nseidle, 2022-08-19:
Oh good golly I've got it all wrong, sorry! Thanks for correcting me. 

I've fixed the docs, again. Please give it a read. In summary, you need the newer `app3M_fat9M_16MB.csv` [file](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/main/Firmware/app3M_fat9M_16MB.csv) and then select the 16M Flash 3MB App partition scheme.

nseidle, 2022-08-19:
Oh, and thanks for trying to compile from source. It's not trivial, and I welcome you to the community of a few who have!

XPav, 2022-08-19:
It's not that I really _wanted_ to, I just needed to get a fix for issue 274 into a unit for a test next week!  And I could!  So it all worked out.

commit 20c6f516 (2022-08-18): Docs to fix #282