sparkfun/SparkFun_RTK_Firmware #285: NTRIP startup is really flaky (timeouts?)

gdt, 2022-08-18:
### Subject of the issue

I realize this report is somewhat vague, but I wanted to get it on the table.

I am observing many NTRIP connections to MaCORS before it finally works.   I did a test run (where I found the bad GGA also), and my log had 37 connections.   2 are bad GGA, and 1 is another issue, which leaves 33 failures and one that worked.

Almost all of the failed ones are very short in duration, and it looks like the ESP just clears the connection state and doesn't respond to incoming data packets.  That's speculation, but what I see is data packets that aren't acked, and that connection never has anything further from the ESP, and then there is a new connection (local port one higher).

### Your workbench

* 2.4RC from August 15
* No bluetooth, and NTRIP to MaCORS over wifi.
* USB to ESP, and full battery

My wifi and internet are very solid; I routinely get > 100 Mb/s speed tests up and down.  The RTK Express is less than 2m optical line of sight to an AP.   I have used MaCORS (Leica network RTK operated by MassDOT) a lot with an Ardusimple F9P, and some with gpsd, and I have not had trouble once I figured out the config.   Not relevant I think, but I'm using a RTCM3MSM_IMAX mountpoint which is network RTK, which gets me into FIX more easily than single base (as one would expect).   So I'm pretty sure I am not having wifi/internet problems.

You can't use MaCORS in CO, but my config is basic rover with enable NTRIP Wifi and send GGA, and I have username/password, IP address of server, port, and mountpoint, and mountpoint pw (?) is empty.   I know these are ok because I get data and RTK float eventually (testing this inside) and I see "200 OK" in the tcpdump.

### Expected behavior

I expect a single connection and for it to stay up, almost always.

### Actual behavior

I don't have the ESP output from this run, but I remember seeing "timeout; closing NTRIP" earlier, I think resulting in tearing down the entire wifi connection and redoing association/DHCP.

I speculate that the timeout for giving up on wifi NTRIP is way too short.
In the source I see 10 seconds.  While I do see that reset on some intermediate success, it feels from watching the logs that it is from the initial startup.

This may be an artifact of "switch to bluetooth" still being there, even though bluetooth and wifi now seem to both operate.  It may also be triggered by the bad GGA/401 issue.

### Discussion

I would suggest changing 10s to 60s for timeouts.   The normal path is that configured wifi will work, and giving up fast doesn't seem useful now that bluetooth runs in parallel.  But at least 10s could be 30s.

nseidle, 2022-08-18:
I've seen this too but because of the changes over the last 10 days I need more info. Thanks for marking the RC you're running. Aug-15 is right in the middle of a series of changes. 

The current theory is that the heap was getting low enough to cause the WiFi connection to become sporadic. I saw this when the heap got as low as ~15k. We've opened up a bit of RAM [here](https://github.com/sparkfun/SparkFun_RTK_Firmware/commit/eb3e4e5d586eff7c58d4a3aa251ed6d5e18f685b) on Aug13 and experimented with ESP32 core v2.0.4, and for my own notes, 2.0.4 uses about 30k more than 2.0.2 so the issue is much more repeatable. For now, RCs are compiled with ESP32 core 2.0.2.

With RC Aug15 as it is, I don't think low heap is your issue. Help me get more information from you.

Things I need from you:
* Turn on heap reporting by going: Main Menu->'s' System -> 'd' Debug -> '2' Enable Heap Reporting
[image]

[image]

* Please log the output generated by the RCs and post in issues. I use TeraTerm to log to a file with [timestamps](https://ttssh2.osdn.jp/manual/4/en/menu/setup-additional.html) turned on.

nseidle, 2022-08-18:
We've also added RSSI output so we can watch that. 

[image]

This setting is also in the debug menu and should be on by default but please confirm it is:
[image]

Obviously at 2m, I doubt RSSI is your issue, but we should watch it all the same.

nseidle, 2022-08-18:
> I would suggest changing 10s to 60s for timeouts.

I don't think this will help.

https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/a880f233ae637662ecf3c87c80ab710e591b1bca/Firmware/RTK_Surveyor/NtripClient.ino#L422-L441

If our connection is dropped, we attempt re-connection. If we don't receive data from the client for `NTRIP_CLIENT_RECEIVE_DATA_TIMEOUT` (10 seconds currently) we give up with error. I assume if WiFi goes bad, the connection is dropped, or we timeout because we have failed 10 times (we get 1 RTCM update per second) to get data from the Caster. Failing 60 times with a 60 second timeout is not likely fix anything. I'm willing to try it, I just need to be able to replicate things.

gdt, 2022-08-18:
What program is best to talk to the console?  nodemcu terminal and kermit both reset the device with DTR, and arduino IDE seems to be display only.   I'm in a mixed Mac/NetBSD environment, no Windows, no proprietary user-space programs beyond macOS.   Obviously esptool knows how so maybe I can just figure out how to tell kemit to not have DTR assserted (which is odd from a traditional RS232 perspective but seems to be the Arduino custom).

nseidle, 2022-08-19:
While Tera Term has the [DTR reset](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/268) reset issue as well, I simply setup my system then let it log. What are you running into other than the initial reset?

gdt, 2022-08-25:
I'm able to talk to the console and will try heap debugging.  But I wonder if this problem is provoked by #303

gdt, 2022-08-26:
After it stabilizes, I'm seeing:
```
Rover Accuracy (m): 0.0141
Batt (103%): Voltage: 4.22V Discharging: 0.00%/hr Green
FreeHeap: 50816 / HeapLowestPoint: 20008 / LargestBlock: 26100
FreeHeap: 50816 / HeapLowestPoint: 20008 / LargestBlock: 26100
[44256650][D][NtripClient.ino:463] ntripClientUpdate(): NTRIP Client pushed 2048 RTCM bytes to ZED
Rover Accuracy (m): 0.0141
[44256770][D][NtripClient.ino:463] ntripClientUpdate(): NTRIP Client pushed 522 RTCM bytes to ZED
[44256981][D][NtripClient.ino:463] ntripClientUpdate(): NTRIP Client pushed 619 RTCM bytes to ZED
FreeHeap: 50816 / HeapLowestPoint: 20008 / LargestBlock: 26100
FreeHeap: 50816 / HeapLowestPoint: 20008 / LargestBlock: 26100
Rover Accuracy (m): 0.0141
FreeHeap: 50816 / HeapLowestPoint: 20008 / LargestBlock: 26100
Batt (103%): Voltage: 4.22V Discharging: 0.00%/hr Green
FreeHeap: 50816 / HeapLowestPoint: 20008 / LargestBlock: 26100
Rover Accuracy (m): 0.0141
```

(need to gather more I know)

gdt, 2022-08-26:
RSSI is -34.  That seems ok :-)

gdt, 2022-08-26:
I just saw a disconnect and while I was not fast enough to cut/paste the last heap line before the disconect was ok and looked like this (same min, similar >50k)
```
FreeHeap: 50948 / HeapLowestPoint: 18044 / LargestBlock: 26100
```

I am starting to think more and more that it makes sense to figure out the SD issue first.

nseidle, 2022-08-26:
I doubt this is related to issue #303. 

> After it stabilizes,

What does this mean?

Ya, I don't think heap is your issue. We are actively working on WiFi stability. Please give [Aug26 RC](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.4RC) a try. We've sorted out the problems between ESP-Now and WiFi. It was causing a lot of AUTH_EXP WiFi failures. We've also added throttling to WiFi connection attempts. If the AP is problematic, it will first give the AP 15s, then 5 minutes to allow connections.

gdt, 2022-08-26:
By stabilize, I mean that by broad experience is a lot of short connections and then there is one that lasts a very long time.

I am using kermit and will figure out logging or find something that will.

I will certainly upgrade to the latest RC.  I don't think I saw AUTH_EXP go by.  I'm 99% sure the AP is fine -- Ubiquiti and lots of other clients with no issues.

Once updated and I have logging, I'll gather logs and tcpdump.    I can also set up an ardusimple with the same caster to be sure that  it's stable, which should be a check that the server and path to it are ok.

nseidle, 2022-08-27:
It's always fun when I get proven wrong.

[image]

Below is a NTRIP Client running with very good RSSI. While I cannot fully explain it, the data is halted for ~10s and then a large amount of data is dumped on the device. This could either be something upstream like the internet/AP/or caster, or it could be downstream (ie, the ESP32 WiFi client is buffering) but I don't think it's the ESP32. Either way, there are instances where data is held for more than 10s. I've increased the timeout to 30s. Increasing to 60s has me worried as the RTK fix will start to fail after ~20s so waiting for 60s is not ideal.

nseidle, 2022-08-27:
With the release of v2.4 I'm going to close this as done so that we can be sure we're all using the same firmware binary. Please open a new issue if you still see problems.

closed 2022-08-27 by nseidle