sparkfun/SparkFun_RTK_Firmware #286: Connecting to bluetooth breaks wifi NTRIP connection

gdt, 2022-08-18:
### Subject of the issue

Again this is a vague report to get it on the table.

I did a test run of booting the Express to watch NTRIP over wifi behavior.   Previous issues are about bad GGA and what I think are overly aggressive timeouts.   After many attempts (35 failures) the connection stayed up.  My memory is that I saw the fix mode icon on the display move to RTK float (indoors, so that's as good as it gets 99% of the time, not a bug) and connected to bluetooth (from QField android).   I didn't notice at the time, but going over the tcpdump the working connection failed, and then the next attempt also succeeded and stayed up through the end of the trace, a bit over 10 minutes.  I was then getting RTK Float status in QField, with position data commensurate with that. (I shut down the Express and then ended data collection.)  The delay from the last NTRIP packet from MaCORS that was acked by the ESP to the first packet of the new connection was 15.

It is of course possible that bluetooth connection setup uses too much heap and causes TCP to break, and then once done the new TCP can work.  But it's also possible that the state machine causes trouble.

### Your workbench
* 2.4RC August 15, Express
* WiFI NTRIP to MaCORS, then bluetooth from QField/Android, which does not send RTCM3 at all.
* Express USB to ESP, full battery, phone on battery (but it is not flaky).

### Steps to reproduce
Wifi NTRIP to MaCORS, let boot and when fixmode goes to RTK (via Express display) connect to bluetooth.   Record all wifi packets with tcpdump.

### Expected behavior
WiFi NTRIP should stay up when bluetooth connects.

### Actual behavior
One out of one times, it didn't stay up.  But I am not 100% sure that the break matched the bluetooth connection time, just 95%.

gdt, 2022-08-18:
More specifically I see
- Normal segments from caster and acks
- after a segment/ack pair, another ack (from the ESP) that is a dup but opens the window (4217 to 5744)
- segments arrive from caster and are not acked
- caster sends R about 8.5 s after last ack
- ESP sends ack 9.5s later (as if it has not seen anything since the dup-ack in point 2) and then a segment with a FIN about 600 ms later, which is only 26 ms more than 10s after the window-opening ack.
- FIN is followed 1.7s later with a "Supervisory, Receiver Not Ready" non-IP frame

nseidle, 2022-08-18:
We've also added RSSI output so we can watch that. 

[image]

This setting is also in the debug menu and should be on by default but please confirm it is:
[image]

Obviously at 2m, I doubt RSSI is your issue, but we should watch it all the same.

nseidle, 2022-08-18:
While I wait for my MaCORS account to activate...

Out of curiosity, do you have ESP-Now enabled? I've modified the issue template to ask that. I do most of my testing with it on as a worst case scenario (WRT RAM).

I'm not able to replicate this. I've connected WiFi + ESP Now + BT, and gotten RTCM via the SparkFun station. I've connected/disconnected multiple times to BT and seen the heap dip to ~45k with little to no memory leaks (heap returns after BT or WiFi disconnect). My WiFi RSSI is -55 (not great, not terrible).

I am operating with the [GGA fix](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/284) in place.

As an aside, I've added the ability to turn off Bluetooth via the System menu:
[image]
If you think BT is your issue, you can now turn it off and re-run your tests.

gdt, 2022-08-18:
I don't even really understand what ESP-Now is and have not tried to turn it on.

nseidle, 2022-08-19:
It's a 'free' point to point radio that is proprietary to the ESP32,. Basically, for short distances, it removes the need for a user to use an external radio (if they are not using WiFi or cellular for RTCM backhaul).

I added Docs here: https://docs.sparkfun.com/SparkFun_RTK_Firmware/menu_radios/

gdt, 2022-08-25:
WIth 18 August, and trying again today, I'm not seeing a problem.  I believe there are issues with the way my device is running that cause wifi/NTRIP to break, but I experienced a BT connect with no loss of wifi, so I think this report was wrong.

I will need to submit something with docs when it really works, but I am able to get RTK fixes in QField with the RTK Express doing NTRIP over WiFi and QField connecting over Bluetooth SPP!

gdt, 2022-08-25:
and a BT disconnect did not cause trouble either.

closed 2022-08-25 by gdt