sparkfun/SparkFun_RTK_Firmware #289: User avatar	Sparkfun facet firmware change request for use in base mode.

amlago, 2022-08-19:
Sparkfun faceta v2.3
I have been using a Sparkfun facet as a base for ntrip corrections, sharing wifi from an lte cell phone, in conjunction with a professional team from another brand. First I locate it with survey in and then I write down the coordinates obtained and to continue working the next day I place it at the base as a fixed base at the same point, where I left a wooden stake and entered the coordinates obtained the day before.
I have been very satisfied.
Because of that I wanted to request a couple of things:
1) if the Wi-Fi signal of the Sparkfun facet is cut, the reconnection time is very short, it is 30 seconds and after that it cannot be activated anymore, to continue using the same base coordinates quickly.
You must then start another survey with survey in and that delays everything.
I would like the Wi-Fi reconnection time to be extended after a disconnection, at least 10 minutes, to be able to fix the Wi-Fi and continue working.

2) I would like that after a mobile base location with survey in, the coordinates obtained automatically pass to the coordinates to be used in a fixed base, so I don't have to enter them manually.

That would be fantastic.

That's it for now, as long as I continue to use Sparkfun facet in daily topography tasks, I will communicate to collaborate within my possibilities.
Greetings and have a great day.

PD: sorry for the Google translation

tonycanike, 2022-08-19:
I can see the value of these two requests, and they would help me.

About point 1, I'd like to use the WiFi to check the coordinates of the fixed base (survey-in or supplied) after the RTCM starts transmitting, without interrupting the RTCM transmission and any logging (OPUS doesn't like gaps in the log).  Whether I use survey-in or if I enter fixed coordinates, I'd like to know where the base thinks it is.  If the WiFi displayed  both ECEF XYZ and LLh coordinates it'd be useful to different people doing different things.  "h" in this case could be the HAE of the antenna phase center, labled as such, so the user knows what's going on.
-If I'm entering manual fixed coordinates, this would be useful to check my entry.  And if I enter h of the mark, HI, and antenna offset, I could see what got calculated for the h of the antenna phase center.
-If I'm using survey-in, I would write it down and use it a couple days later to correct all my rover point coordinates when I eventually get an OPUS solution.  

About point 2, perhaps the code could save the survey-in coordinates in a settings file, so the next day the coordinates are retrieved.  Save them in both ECEF XYZ and LLh so someone could adjust the h for a different HI (rod/tripod height) the next day.

amlago, 2022-08-19:
Good, you understood me perfectly.
I wish it was possible, it would help me a lot in my daily work.
I am a Surveying Engineer and I loved having my own base for so little money.
Cheers

amlago, 2022-08-19:
I forgot to tell you that I use a professional controller to save data and it informs me of the coordinates of the base station.
But it would be much more useful for me if they were directly in the Sparkfun facet.

tonycanike, 2022-08-19:
I use SurvPC on my data collector and I'm trying get it to SET the base coordinates.  Still working on that.

nseidle, 2022-08-25:
> if the Wi-Fi signal of the Sparkfun facet is cut, the reconnection time is very short, it is 30 seconds and after that it cannot be activated anymore, to continue using the same base coordinates [...]

Are you using WiFi for NTRIP Server mode? If yes, I believe we will have this fixed in the next release. We've added throttling to add increasing time between attempts when the connection to the NTRIP Caster fails. The first attempted connection is 0 seconds (unit was just turned on). The 1st failed attempt has a 30 second delay. After that, the attempts increase by 5 minutes between connection attempts (5, then 10, 15, 20, etc). This is meant to avoid getting rejected by Emild casting service, and banned by RTK2Go when too many reconnection attempts happen too closely after a connection is lost.

> I would like that after a mobile base location with survey in, the coordinates obtained automatically pass to the coordinates to be used in a fixed base, so I don't have to enter them manually.

Agreed. Moved to #302.

amlago, 2022-08-25:
Yes, I use wifi from a cell phone that I leave next to the Facet. I did not know that it repeated the attempts every 5 minutes, so I will do simulation tests of wifi cuts in order to stay calm.
What Rrtk2go does with multiple failed attempts has happened to me and they block your IP.
Touch restart the router or cell phone.
I clarify that the few times I have had connection problems, they derive from my cell phone with LTE and not from Facet.

amlago, 2022-08-25:
I actually thought that after 30 seconds, it was no longer trying to connect; I didn't wait long enough.
Thanks for the clarification.

nseidle, 2022-09-08:
I believe the issue "Facet fails to reconnect to cellular AP WiFi if connection is lost" has been fixed in v2.5. Please give the release candidate and try and let us know.

amlago, 2022-09-08:
I will test v2.5 and comment on the results.
Thank you

amlago, 2022-09-09:
I tried v2.5. Changes are seen on the screen with respect to the v2.3 that I used.
You can connect, for example, the SWmaps by bluetooth at the same time as the RTCM sending, to see, for example, the base coordinates, but it seemed to me that the sending slows down. At least the speed of the count on the screen.
Regarding the wifi disconnection, it remains the same. After the programmed time, the Facet no longer tries to reconnect to wifi and I have no way to do it manually, except to start the whole procedure.

amlago, 2022-09-19:
With v2.  5 9/19 got Facet to connect to wifi automatically after accidental disconnection.  After 15 seconds without problem and then after 5 and 10 minutes it also connected correctly.
Excelent

nseidle, 2022-09-20:
Excellent! Thanks for testing.

nseidle, 2022-09-21:
This has been fixed in [v2.5](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.5). Please open a new issue if the problem persists.

closed 2022-09-21 by nseidle