sparkfun/SparkFun_RTK_Firmware #290: Firmware upgrades overwrite custom profiles settings

tonycanike, 2022-08-19:
### Subject of the issue
I've setup a few different profiles for various logging configurations and fixed base locations.   Firmware upgrades appear to overwrite the custom profile settings. 

### Your workbench
* RTK Firmware RC Aug 18
*  No radios

### Steps to reproduce

1. Setup a few profiles.  
2. Check the settings files (e.g. see the profile name in the setting file).  
3. See your happy profile names on the display when selecting profiles.
4. Use a setting file and be happy.
5. Upgrade firmware.  
6. Try to select a profile and don't see your happy profile names.
7. Check settings files, profile name is reset to a default name (e.g Profile2).
8. Be less happy.
9. Remember you backed them up on your PC!!   woo life is gonna be good again!
10. copy to SD card.
11. Start Facet.
12. No happy profile names.

### Expected behavior
If I copy my settings files to the Facet SD card from my PC, and the filesizes/FW version doesn't match, the software should read in what it can, fill in the blanks/errors with defaults, and write the file back out.  The user could then diff the file on the SD card with a file they've saved elsewhere to see what changed and what they need to fix.

### Actual behavior
Firmware upgrades overwrite settings files with default settings. If I copy the settings files to another computer and copy them back to the Facet SD card after an upgrade, it seems that they are getting overwitten too.  There's a message about expected settings file size difference and the files being rewritten.

nseidle, 2022-09-01:
As usual, thanks for reporting! And making me grin - happy profiles!

The detailed list is much appreciated but I cannot replicate this yet. 

The user setting files live in LittleFS; this is a chunk of flash inside the ESP32. We don't touch this with two exceptions:

* "Factory reset" formats the entirety of LittleFS erasing all user settings
* Using esptool to do a erase_flash (only the devs do this, and only in rare occasions when things really hit the fan)

> 5. Upgrade firmware.

How did you upgrade the firmware? I just used the GUI and the SD path to upgrade an RTK Express any my profiles stayed intact.

>  If I copy the settings files to another computer and copy them back to the Facet SD card after an upgrade, it seems that they are getting overwitten too. There's a message about expected settings file size difference and the files being rewritten.

The SD settings *should* override the internal system settings.

After firmware update, if the settings file don't match the system *does* read in what we can. Below is the output I see:

    Warning: Settings size is 4496 but current firmware expects 4488. Attempting to use settings from file.
    Unknown setting enableRtcmMessageChecking
    Unknown setting bluetoothRadioType

Note: Settings files have the platform name in them. So if you're moving settings from an Express to a Facet, for example, you'll have to rename the file.

nseidle, 2022-09-01:
Ok, I think I found the problem. [This commit](https://github.com/sparkfun/SparkFun_RTK_Firmware/commit/262ab2fe3afbfb95c381bc9351ac5999a9bfca0a) was forcing the LFS settings onto the SD card. So, if you:

* Factory reset device
* Started device (Profile1 would be created)
* Powered down unit
* Inserted a card with a profile 'HelloWorld'
* Powered on unit
* Unit would overwrite HelloWorld at beginBoard() because of the enableResetDisplay feature

I think I've got it fixed in next [RC v2.5](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.5RC).

tonycanike, 2022-09-02:
Cool, thanks Nathan!  I was trying to replicate another unrelated issue ( #292 )  when I noticed the above behavior.  Looking back at my write up for #292,  I use multiple different firmware versions (via the SD card and serial menu) and I was factory resetting to ensure that I didn't have something misconfigured that was causing the behavior I saw in #292.  So your results make sense to me.

nseidle, 2022-09-21:
I believe this has been fixed in [v2.5](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.5). Please open a new issue if the problem persists.

closed 2022-09-21 by nseidle