sparkfun/SparkFun_RTK_Firmware #291: Possible rearrangement of the menus when turning on?

amlago, 2022-08-19:
Sparkfun facet v2.3
Every morning I have to make some modification in the coordinates of the base to work.
-Is it possible to rearrange the menus on my own so that the Wi-Fi configuration is in the first order? or the menu that interests me according to the needs of the job?
-That way some users would save touches of the button, lengthening its useful life.
Tell me if it is possible or useful for some users.
Cheers

nseidle, 2022-08-22:
Hi amlago,

Good feedback. Have you checked out the [profiles menu](https://docs.sparkfun.com/SparkFun_RTK_Firmware/configure_profiles/)? You should be able to create up to 8 specific configurations that can be switched with a few button presses. Please re-open this issue or provide more feedback if you

closed 2022-08-22 by nseidle

amlago, 2022-08-22:
Hello, the profiles are very useful and I use them but I was thinking of something else.
I was interested to know if it was possible to reorder the menus on my own so that the configuration item would be first, for example.
Sometimes in the field you have to enter data that arises at the moment and it could be useful for me if the first item was "configuration".
And sometimes it would prevent button touches and prolong its life.
I understand that these changes would be personal, so I was asking if it would be possible for me to reorder them.

Thanks for answering.