sparkfun/SparkFun_RTK_Firmware #292: RTCM Xmitting "Off" displayed in RTK Base-Rover external radio configuration

tonycanike, 2022-08-22:
### Subject of the issue
RTCM Xmitting displays as "Off" after Base Survey-in completes

### Your workbench
RTK Firmware RC August 18
uBlox firmware 1.32.
Facet set for RTK Base
Holybro external radio

* Are there any additional details that may help us help you?
First attempt to use Base-Rover after upgrading from firmware 2.1 and I saw this issue.  I've been doing a bit of static PPP processing and haven't been using the Base-Rover external radio RTK configuration since early July.

I reverted to 2.1 and was able to see RTCM Xmitting counts increasing on the display.  I had tried v2.3 and v2.2 with no luck. But after I loaded v2.1, I remembered the bug where you had to power up, select Base,  let the base survey-in, see 0 on the RTCM count, power cycle, and start all over and then it shows increasing RTCM counts (Nathan you and I discussed this behavior months ago.)

For the record, the double start-base power cycle trick did not work...still see "Xmitting Off" on the display.

I reloaded RC August 18 and saw the same "Xmitting Off" display,

### Steps to reproduce
Use serial menu to reset to factory defaults.
See "Factory Reset" on screen.
Device boots into Rover mode.
Select "Base"
Wait for Survey-in to complete.
Base displays "Xmitting Off"
Red data light blinks on Holybro.
I don't have my Rover setup so don't know what's being transmitted.

I repeated the above with the 18-Aug RC four different times; very repeatable.

### Expected behavior
Base should display "Xmitting"   with increasing RTCM counts

### Actual behavior
Base displays "Xmitting" with "Off" instead of the RTCM counts.

nseidle, 2022-08-23:
Yup. Should be fixed in next RC.

RTCM has gone from being transmitted over external radio only, to four targets: external radio, WiFi NTRIP Server, ESP-Now, and Bluetooth. I think we've got it in a better place now.

nseidle, 2022-08-27:
This is fixed in v2.4.

closed 2022-08-27 by nseidle