sparkfun/SparkFun_RTK_Firmware #302: Automatically populate Base Fix ECEF after Survey-In

nseidle, 2022-08-25:
After a Survey-In, automatically populate the Fixed Base coordinates with the obtained position.

I like this idea for a few reasons:

* Using a Facet L-Band, you can get a ~20mm fix, then survey in, then have a 'relay' type base station that has absolute accuracy, that only takes 60s to setup.
* It allows a user to physically mark a spot, survey in on top, then return over multiple days and have the surveys all be relatively accurate to each other.

My concern is how to make the user interface as easy as possible.

How it would work:

* User presses Setup to start Base (preconfigured for Survey In)
* Survey-In completes, stores/copies coordinates into Base ECEF
* Next time user presses Setup and starts Base, it would be Fixed at previous coordinates

The problem is that the next time the user presses the Setup button to start a base, should we start in the temp base or the fixed base mode? We want fixed base. But then how would the user easily indicate they actually wanted to start a new survey-in at a new site? They would need to open up the config via WiFi or Serial and reset the 'do a survey-in next time'. This seems cumbersome. 

How could you re-trigger the Survey-In to Fixed Base setup?

* We could add an 'E-Base' menu below 'Base' on the display. This extra menu would only be seen if the user enabled this specific system. 

If anyone else has any ideas, I welcome them.

gdt, 2022-08-25:
My quick reaction is that the complexity of what is possible and might be here in 6 months is pretty big, and that trying to put it all into two buttons and that little display (which are great compared to none!) is too much.   In particular, for the workflow of establishing a mark, survey-in, and using it as a base, I can't see anybody wanting to do it with just the buttons.  I should warn you that I'm reading an interaction design book "About Face" by Cooper et al. so I may be a little odd.  But seriously I suggest that you read it, and don't be put off by part I.

FIrst, I feel like I'm missing something, because the idea seems to be to use satellite-transmitted RTK data (misnamed by the upstream data provider as L band :-( ) to get a more accurate base location.   Is it possible for "base" to be somehow "rover"?   Or is this really "operate to determine position for storing" and is neither base nor rover?

I also do not understand Facet vs non-Facet.   As I understand it, that's just form factor, and either way you need to put the antenna over the mark (and if you use the same HI that's ok, but for rigor one would want to record HI).

I'll observe that reference data could be from "L band" or from NTRIP from someplace else.    As I see it there is nothing magic about "L band", only that the data link works even in places without cell service and the reference data is valid in a huge range.   In my case, MassDOT's MaCORS is excellent, but I might get better RTK fix behavior under trees with a base only hundreds of m away vs the network base (I have no idea really, but interesting to try), so using MaCORS to establish a local base and then using it would be exactly the same concept as what you're proposing, but with a different reference source.

As to the actual user requirements, I'd want to be able to (this is trying to be needs not UI):
  - physically set up the antenna over  the mark, and record HI and antenna serial number
  - tell the unit, somehow, to start a survey-in with {external RTK source} and set the obtained coordinates
  - record somehow, in a way that is not subject to getting lost, those coordinates, in addition to them being stored
  - start as base-with-coords
  - tomorrow, re-start as base-with-coords **with no possibility of messing up the previous coordinates**
  - perhaps, when starting, re-do the survey in to get a delta from the stored one, and display the delta, record this info but use the stored coordinates from the setup
  - setting up the base the first time and later involves a prism pole with tripod, or something, and is going to be a few minutes.  So the computer part doesn't need to be 10s, but 1 minute would be nice.

Which leads me to say, don't put this in the buttons at all; I can't see how that can be without leading to risk of error.  Instead, use wifi config, for now, with a menu/button to do this, and resulting in a display of the ECEF and whatever uncertainty estimates.  This should all be written to uSD, but the user could screenshot this, as a backup in case of trouble.

amlago, 2022-08-25:
I like the idea in general and proposed it in #289.
I would not like menus or extra functionality to be added to the power button.
**Simply that the coordinates obtained in survey in pass to the fixed position to use them the next day.
If you would like other coordinates or verify that they are correct, you enter through Wi-Fi configuration and it is ready.
I would like that which does not require adding functions to the power button.**
I remember that by default my Facet has some ecef values in the fixed base, then it would be necessary to replace those values with the last ones obtained with survey in.
Nothing else and it's perfect.
I tell you that in the daily work in surveying using Sparkfun Facet as a base, I must always use wifi configuration for the base settings.
In general, I don't care about the great precision of the base coordinates, but that they are repeatable the next day.
The base in the field is marked with a concrete or wooden monument and when parking on it every time, it is enough for me to know the first coordinates obtained with survey in and to be able to repeat them.
So, just get fixed base autosave and that's it.

tonycanike, 2022-08-25:
Lots of good thoughts above from everyone.  As @gdt discussed, the UI for additional features gets challenging because the UI is a small screen and one button.  It also "just works" for the simple use cases for people needing approximate locations.  I'd guess a lot of folks buy these units, they are still learning all the complexities, and want the units to "just work."

I have found entering fixed base coordinates with the WiFi to be challenging when out in the field, so changes to make this easier and less error-prone are welcome.  Being able to reuse coordinates is a very useful improvement.

Just seeing the OLED on the Facet to put it in WiFi mode is challenging when the mosquitos are biting, the sun is very bright, and the Facet is 3" above eye level.   Then I need to get the phone connected, and enter the long strings of numbers.  Oh yes also need to do math to adjust for the HI and antenna offset.  I'm never quite sure I got it right.  A bit of patience, experience, and discipline help here, but it's still error-prone.  So anything that simplifies this (also see #269) would be a big help.

What I see my professional surveyor friends doing is using their data collectors (DC) to save many points in their DC.  They load OPUS solutions as points into their DC. The DC points contain the coordinates of the mark, not the antenna.  The DC knows the antenna offset (a setting that's saved.) When they save a new point in the DC, they enter the current HI and a short text description.  When they setup the base on a mark on a new day, they enter the today's HI and choose a point from their DC to set the base.

(The careful folks save important points in their DC and also write them down on paper.  Then they reuse the points from the DC another day and double check the values displayed by the GNSS unit against their paper notes.  They write down the different HI values each day to verify h got set correctly each day.  Some of them measure HI both in meters and in feet and write both down on paper, that's how they find typos back in the office when things don't look right.)

Could the marked points file and mark button be used to implement something similar?   There's already a mark button on the menu.  There's already a CSV file with marked points in it.  I've never used the mark feature or looked at the file format, but there are a small number of surveyor standard CSV survey point file formats (eg:  pt#, northing, easting, height, description) in use today that just about every surveyor understands.  Northing, easing, and height can be in many different datums and the surveyor just needs to know.  Almost like a mini data collector built-in to the RTK Firmware.

Profiles also contain base coordinates - could base coordinates be saved and reloaded from the profiles?  We could have a long working session about the profiles, great feature and a lot more could be done with them. But that's another topic.

As @amlago said, it's often most important that the work - the data collected - over a few days is relatively precise, and not necessary accurate in the world.  Setting the base over the same mark multiple days in a row with and having the base use the same coordinates each day, even if they are inaccurate by many meters to the real world, is a very useful feature. 

It's very likely that the **base receiver antenna will be at a different height over the mark each day, so being able to account for different HIs on different days is important.**  Sturdy tripods are less likely to get blown over in the wind than a fixed height pole with a bipod, so many surveyors prefer tripods for the base.  Though every day a tripod is going to be at a different height, producing a different HI every day.

It's common to spend a couple days RTK'ing a site, with good internal precision using a base with an approximate survey-in location.  While the surveyor is out with the rover, the base is also collecting raw observations.  Then back in the office a week later the raw obs are used to get an OPUS solution, and all the data is then adjusted to be accurate in the world.   So using the same coordinates (adjusted for different HI) each day is big help.
  
@nseidle restricting this to ECEF means the base needs to have the exact same HI on successive days.  This often isn't practical.   LLh (Lat/Lon/Ellipsoid Height) allows the incorporation of various HIs on various days.  As @gdt once commented, over the height the antenna over the mark (~2m or so), the difference between the geoid perpendicular vector and the ellipsoid perpendicular vector is very very small.

amlago, 2022-08-25:
Analyzing the comments, I see the benefits of keeping the idea but saving the LLH coordinates and putting the possibility of entering HI.
I liked that idea and for my purposes it would work fine.
The HI option could be left with the initial value zero (0) in case I don't mind correcting the height value of the mark. I would only change it if necessary. I think it would be an intermediate solution.
I hope you have understood, because my native language is not English.
Go ahead with the idea if there is consensus.

tonycanike, 2022-09-03:
> Simply that the coordinates obtained in survey in pass to the fixed position to use them the next day.
> If you would like other coordinates or verify that they are correct, you enter through Wi-Fi configuration and it is ready.
> I would like that which does not require adding functions to the power button.
> So, just get fixed base autosave and that's it.

@amlago I'm not sure this is the best UI.   If this is implemented, then you are *requiring* a user who wants to have fresh survey-in base coordinates to use the WiFi configuration.  This is a lot of extra time, and requires hardware (a phone) that a user may not have or want to use. 

I believe I'm agreeing with  @nseidle here:

> The problem is that the next time the user presses the Setup button to start a base, should we start in the temp base or the fixed base mode? We want fixed base. But then how would the user easily [...]

Here's one idea:   

1.  The main setup (OLED) menu is changed to have two base options
a. Base Survey-In
b. Base From Profile 

2. If option 1a "Base Survey-In" selected, perform survey-in procedure, set the ZED base coordinates, and go.

3. If option 1b "Base From Profile" selected, load coordinates from the current profile.  Use ECEF or LLh as defined in the profile.  Set the ZED base coordinates and go.
3a.  Allow user to use WiFi configuration to view coordinates.
3b.  Allow user to use WiFi configuration to enter today's HI if LLh.   Reset ZED base coordinates.

4.  Base coordinates are auto-saved in the current profile:
a.  If the base coordinates are determined by Survey-in, they are saved in the profile as ECEF.
b.  If the base coordinates are set by WiFi configuration, they are set as entered (ECEF or LLh) by the user.
c.  If the base coordinates are set by Serial configuration, they are set as entered (ECEF or LLh) by the user.

5.  User can switch profiles to access a handful of base coordinates preloaded.

6.  User can edit profiles to pre-populate a small number of base coordinates.  User can specifcfacet

@gdt made an important point about the UI.  At some point the ability to load functionality into the one button will diminish...if it hasn't already.   Is it time for a custom app?  A custom app would be less cumbersome than the WiFi AP Browser configuration.  It's a huge amount of work though.

Here's how emlid does this:

https://www.youtube.com/watch?v=p7L-3E-OM-c

tonycanike, 2022-09-04:
Eh I now see the mistake in my "Base from Profile" menu name suggestion. This menu could be to use coordinates from the Wifi config or serial menu. And the name is too long for the screen.  

Another issue with this menu option functionality is if the user never entered any coordinates and this is their first time, the base might use fixed coordinates for SparkFun HQ.  

Or it might use fixed coordinates for the last location the base was used, which could be very far away.  This violates the "should just work" principle.  

Somewhere I've read about a GNSS receiver that warns if the entered fixed base coordinates are "too far" from the current location the GNSS receiver has determined for itself.  

If it's "too far", what does the RTK device then do 
- just use the very wrong coordinates? Would lead to erroneous rover coordinates 
- revert to survey in? Could ruin a day's work for someone who was expecting to use fixed coordinates. 
- revert to being a rover? Might be the best, simply refuse to 'just work' if it can't 'just work. Describe this behavior in the docs. 

And how far is "too far"?   I've read about folks who want, for example, NAD27 coordinates from their rover so they give the base the NAD27 coordinates of the base location. WAnd how far is too far on the vertical?  I could see someone reasonably wanting to enter orthometric height for example ("I entered the elevation on the brass disk.").  These 'hacks' kinda work in small areas.

gdt, 2022-09-04:
Wow, I am surprised about NAD27 and orthometric height.   I agree about "kind of work", but these seem beyond what's reasonable, tens of meters differences.

I must admit that I've always been uncomfortable with survey-in as a concept, but I guess there's the idea that getting 1-2m with clear sky and replicating that relative error precisely is useful, and maybe there's a standard workflow to do OPUS/PPP from the base later and adjust the rover points.

I think I lean to "if requested fixed coordinates are more than 10m different from a 1-minute average, stop and throw an error", but I guess that could be hostile to the NAD27 people, so maybe 50m or 100m?  Beyond that, I think overwhelmingly it's a blunder.

tonycanike, 2022-09-05:
100m is probably reasonable for the allowed distance between entered fixed base coordinates and the receivers' current approximate position.  It could always be changed later if there's a problem, or it even could be a configuration setting.

I found these two comments on a Trimble Access help page. Trimble Access is their data collector software.  So Trimble says they use 300 meters before a warning is issued by their data collector software.  I do not know if the firmware in their GNSS receivers has this functionality.

`If the keyed in coordinates differ from the current autonomous position generated by the receiver by more than 300 m, a warning message appears.`

`The coordinates you enter should be as accurate as possible. Every 10 m of error in a base station coordinate can introduce a scale error on every measured baseline of up to 1 ppm.`

https://help.trimblegeospatial.com/TrimbleAccess/latest/en/GNSS-base-coordinates.htm

And yes, exactly, there is a workflow to survey-in base coordinates, start gathering raw base obs, and spend the day surveying to gather rover points.  A few days later, back at the office, get better base coordinates and adjust all the rover points.   This allows the field work be performed with one site visit,  Otherwise it would require two site visits separated by a few days to get a good OPUS solution for the base first, and then gather rover points later.

gdt, 2022-09-05:
So there are two concerns:
  1) avoid scale error from being too far off even though you will correct.   100m and 300m are close
  2) avoid blunders in setting up with wrong coords

nseidle, 2022-09-08:
How does this look?

[image]

Antenna height and ARP are added. And a button to copy current LLH into the fixed boxes.

As gdt points out, adding more to the display menu is probably folly. I think this is the simplest solution but welcome feedback.

I was going to order "About Face" by Cooper but, wow, 1200 pages I cannot handle at the moment. Please continue to guide us in the mean time.

'Use Current Coordinates' button is also available for ECEF:
[image]

As noted in #322, the LLH/ECEF coordinates within the AP page are not (yet) dynamic. I assume users will use AP to survey in, then re-connect to AP config to switch to Fixed, and hit the 'copy' button. By definition of Survey in, LLH/ECEF don't change so it doesn't need to be dynamic.

tonycanike, 2022-09-08:
Looking good!

1. **Clarify current coordinates (what are they**):   
Are the coordinates displayed up in the header of the web page the coordinates of the antenna phase center directly from the F9P?  Someone might think they are the coordinates of the mark the pole tip is on.   Maybe there could be a label so there is no chance of confusion.  
How about a putting "(APC)" right after the height number on the LLh line because that's what really needs to be clarified.

2. **Need to adjust h when copying LLh**:  
If you want to fix the base to the current LLh coordinates by pushing the "Use Current Coordinates", and the pole height and offset are non-zero, the HAE needs to be be adjusted by the pole length and antenna offset.

That is, if the HAE in the header is 1555.643 (APC), if the pole height and/or offset is non-zero, and if you want to set the base to those coordinates,  then the HAE in the Geodetic Coordinate section of the Fixed Base configuration should be 1553,5919   (1555.642 - 2.000 - 0.0421). 

To say it another way, if someone enters a non-zero height and offset, then they are entering the coordinates of the mark (pole tip) and not the antenna phase center.

If the height and offset are zero, then they are entering the coordinates of the antenna phase center.

amlago, 2022-09-08:
The presentation of data on the configuration page I like very much. I find the initial location data very useful.
This is not exactly what I requested, as @tonycanike  said, but I would be satisfied if it works for other users.
This would be a path similar to what I have proposed, with some improvements.
With regard to pole heights and phase center values, users who work professionally or semi-professionally in surveying will have no problems or confusion. With respect to the casual user I could not say anything.
The ideal would be to be clear about the phase center values and directly compensate them by software without action by the user, depending on what type of Sparkfun rtk you are using.
It would only be necessary to enter the height values of the pole or tripod as you have.

nseidle, 2022-09-08:
@tonycanike:

1) How's this?
[image]

2) Need to adjust h when copying LLh:

Wait, do we tho? Let's say I'm setting... (does scribbling on paper). Ah, I see your point. User surveys-in a point, then the copy button *could* subtract off the HI so that the 'fixed' location removes the variables of the HI. I feel like modifying the HAE number is dangerous. I worry about a user who does a survey-in, then fixed based setup as we describe. Then returns the following day, sets up, unit is in fixed based, user opens AP, adjusts their antenna height to today's prism pole, and then hits 'Use Current Coordinates' again. The HAE will be (in this example) 2.0421m too low.

3) @amlago - I'm still trying to work in what you want without creating an 'amlago' button :)

> I would like that after a mobile base location with survey in, the coordinates obtained automatically pass to the coordinates to be used in a fixed base, so I don't have to enter them manually.

Couldn't profiles be used to get this? 

A. User enters AP config and sets profile to 'Joe's Farm NE Corner'
B. User sets up over point and does a survey-in
C. User goes into AP config. Changes from Survey In to Fixed.
D. User enters Antenna Height and ARP
E. User presses 'Use Current Coordinates'
F. User saves profile

The base will begin transmitting RTCM fixed off the HAE of the mark on the ground.

Days later, user should be able to load Joe's Farm profile:

A. User enters AP config and sets profile to 'Joe's Farm NE Corner'
B. User sets up over point
C. User goes into AP config and verifies Fixed geodetic coordinates
D. User enters any new Antenna Height and ARP
F. User saves profile

The base will begin transmitting RTCM fixed off the HAE of the mark on the ground.

Does that make your job easier or harder?

amlago, 2022-09-08:
I don't see the profiles option to save coordinates as useful.
It leaves me very satisfied with what was raised in the images above.
With that I would work without problems and as a topography professional every time I change base or task I am very careful with the height of the cane or the height of the tripod.
That configuration raised in images above is sufficient for my needs.
Thank you

tonycanike, 2022-09-09:
1.  Yes, thanks!

2.  Hmm.  Good question.  Let me step back some and think it through. 

I thought the original request was to
(a) save the coordinates of the base one day. 
(b) reuse the coordinates for the base the next day, with a different antenna height, perhaps even a different antenna with different offset. 

Since the next day might have a different rod height, doesn’t that imply that step (a) needs to save the mark coordinates?  So today’s antenna height and offset need to be subtracted from the HAE from the F9P-supplied coordinates at some point before saving (to the profile?). 

So now we have 
(a) save the mark coordinates (in the profile?) of the base one day for use tomorrow. 
(b) retrieve the mark coordinates (from the profile?) for the base the next day, with a different antenna height, perhaps even a different antenna with different offset.  

So in step (b), the code will need to add the antenna height and the offset to the mark coordinates before configuring the F9P.  After configuring the F9P, the screen shows the APC coordinates in the header…and the mark HAE in the form field. 

In my mind, the main utility of the ‘use current coordinates’ button after a survey in is so that a user can enter antenna height and phase center so the coordinates of the mark can be calculated and and displayed so the user can write them down   and so the coordinates of the mark can be saved in step (a) above.  

And step (b) needs to retrieve the mark coordinates, display them, allow the user to enter antenna height and offset, add the three height numbers, and set the F9P. 

I think the HAE entry field for the geodetic fixed base coordinates should consistently display or be used to enter the mark coordinates. And be labeled as mark HAR. 

Surveyors work with mark coordinates, so they’d want to enter the HAE of the mark into the screen  If they surveyed in or reused yesterday’s coordinates, they’d want to read the HAE of the mark on the screen so they can check it and write it down. 

In all of the above, if the antenna height and offset entered by the user are zero, than the phase center coordinates and the mark coordinates are the same.

tonycanike, 2022-09-09:
The profiles already save fixed base coordinates in them - what was the intent of that?

amlago, 2022-09-09:
Hello, I will clarify my position on the subject.

1) Originally request that the coordinates obtained with survey in will automatically go to the fixed base position.

2) At the same time it was requested to add rod heights and phase center offset.
3) What was proposed as a solution for both things was what was proposed in images by @nseidle and it seemed very good to me.

So in summary I would like to leave the wifi configuration raised in images before.
 It seems to me that for daily use, where I use approximate bases, monumented with wooden stakes, it is enough and I liked the idea of ​​starting to work with a base in a fixed input mode.

Base Mode: Fixed
Type of coordinates: geodesic
Use current coordinates button
Enter rod heights and phase center offset. (be careful with the value and sign depending on whether the configuration is programmed and whether you want to obtain mark heights or not)
save and work

 
Suggestions:
a) Write down on paper the coordinates used, the height of the rod and the phase center shift just in case. ha
b) If necessary, do a post process and correct the approximate base coordinates. In general, this is not necessary for my work in agriculture and irrigation.
Cheers

amlago, 2022-09-09:
I forgot. Sending fixes with fixed base ntrip, it seems that message 1005 is not being sent. At least Rtk2go doesn't show it to me. With previous versions it looks perfect.

amlago, 2022-09-10:
Suggestion.
The default values of the base coordinates, which I believe are the ones corresponding to the Sparkfun Company location, should then be replaced by the values obtained at the current base location.
In other words, if a base location was never obtained, the coordinates will be the default coordinates, but when locating a base, its coordinates should replace the default ones.
Hope it's understandable.

nseidle, 2022-09-16:
Moving [comment](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/302#issuecomment-1242728343) to separate issue [here](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/326) so that we can better track individual requests and issues.

nseidle, 2022-09-21:
I believe this has been implemented in [v2.5](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.5). Please open a new issue if there are additional features to discuss.

closed 2022-09-21 by nseidle