sparkfun/SparkFun_RTK_Firmware #303: firmware complains about uSD filesystem but it works

gdt, 2022-08-25:
### Subject of the issue
Sorry, github just ate my issue that I spent a lot of time on so this is too brief.

I am seeing complaints about the uSD on the ESP serial port, that it is invalid and can't be opened.  These seem to be about the filesystem format.

### Your workbench
* RTK Express
* Lexar 1000x 32G uSD, formatted with Disk Utility on a mac
* using wifi only with NTRIP

### Steps to reproduce
copy settings off, format with Disk Utility, and copy settings.txt and setttings_0.txt back.  Remove the mac good: spotlight and .fseventsd, and eject.   Put it in and boot up the RTK Express and watch the serial port.

### Expected behavior
uSD should be accessed with no issues

### Actual behavior
Lots of messages about "can't init" the SD.  Mostly it looks like
```
1332909][D][Begin.ino:230] beginSD(): SD card detected
[1334910][D][Begin.ino:244] beginSD(): SD init failed. Trying again 1 out of 1
SD init failed. Is card present? Formatted?
[1337265][D][NVM.ino:137] recordSystemSettingsToFileLFS(): Settings recorded to LittleFS: /SFE_Express_Settings_0.txt
```
But yet it does RTK, and that makes me wonder if the settings are also in the flash -- but I have not figured this out from the docs.

### Requests

1. Could someone else test with a Mac-formatted uSD?  A theory is that the presence of the mac goop directories -- which should just be ignored -- is causing trouble, but the other theory is that it's something else.
2. The code should emit better diagnostics even if it's a codepoint I have to grep for in the sources.

nseidle, 2022-08-26:
> Could someone else test with a Mac-formatted uSD? 

I suspect it's Mac + 32GB limit but I can't test. The max size our SdFat implementation supports is 32GB. I haven't, and currently can't, test with OSX formatted cards. Thank you for testing! Please continue to dig and let me know if we need to update any docs.

> The code should emit better diagnostics even if it's a codepoint I have to grep for in the sources.

    [1334910][D][Begin.ino:244] beginSD(): SD init failed. Trying again 1 out of 1

Well, it's line 244 of Begin.ino inside the beginSD() function. How much more do you need? :) But in all seriousness, the firmware is unable to mount/read the card as formatted. 32GB should work but consider moving down to a 16GB card. If you really need larger card support, we can go there but on a different issue.

> But yet it does RTK, and that makes me wonder if the settings are also in the flash -- but I have not figured this out from the docs.

[Indeed](https://docs.sparkfun.com/SparkFun_RTK_Firmware/system_configuration/).

All device settings are stored both in internal memory and an SD card if one is detected. The device will load the latest settings at each power on. If there is a discrepancy between the internal settings and a settings file then the settings file will be used. This allows a collection of RTK products to be identically configured using one 'golden' settings file loaded onto an SD card.

gdt, 2022-08-26:
I don't need 32G; that's just what I had handy.  I didn't realize that was near any limit.

The unit has in the past written settings and logfiles to this card.   So I'll reformat on NetBSD and see how that goes.   I suspect a reaction to mac goop, and if that's it I'll close this and open a clean new issue, as eventually that ought to work.  Easy enough for me to work around it.

gdt, 2022-09-17:
I mounted the uSD on NetBSD, removed /.metadata_never_index, and removed the SFE settings  and backup files.  I then unounted and ran fsck_msdos which objected to and cleaned up a few things.   I put the card in the Express and booted it up in rover mode for a few hours and got a fresh config file stored and a ".ubx" file with NMEA in it.  This is the same 32G card.

So it looks like this is a mac issue.  I'll open a new issue, and try to do some better testing.

closed 2022-09-17 by gdt