sparkfun/SparkFun_RTK_Firmware #305: serial config menu exit timeout is very much too short

gdt, 2022-08-26:
### Subject of the issue
I'm trying to gather debug information, and deal with the config menus.   I find that in some short period of time, maybe 30s, it gives up on input and returns to the previous.   I'm the human and I'm telling the device what to do, so it should do what I ask.  I realize that after some really long amount of time, it makes sense to assume "human has left", but that should be so long that it is almost never the case that a human looking at the display is annoyed by an unrequested menu exit.

### Your workbench
* 2.4 August 18, WiFi running NTRIP, kermit on ESP config port

### Steps to reproduce
Type a key for menu, type s, type d.  Start to read the huge menu (it's ok that it's huge) and also try to read the debug messages that were printed above the menu and think about them.  See it exit the menu and scroll when you did not ask for that, and then exit the next menu.

### Expected behavior
User requests should be respected.  I suggest changing the timeout to 10 minutes.  If you walk away, it doesn't matter what happens, and if you are there you want it to do what you asked it to do.

nseidle, 2022-08-26:
Yep - I agree. I need to finish out v2.4 but we'll get this in soon.

tonycanike, 2022-08-26:
Agreed!  Increasing the time out will help. Thank you.

commit 5564427e (2022-09-09): Increase all menu timeouts to 10minutes

Fix #305
closed 2022-09-21 by nseidle