sparkfun/SparkFun_RTK_Firmware #306: Be able to load, modify, and save profiles without rebooting. Don't auto save custom profiles.

tonycanike, 2022-08-26:
It would be useful to be able to load, modify, and save various custom profiles without rebooting. If you enter the profile serial menu, load a profile, and then leave the profile menu to change a setting, the firmware reboots before you can change anything.  

It would be useful to load a custom
profile, tweak it, run with it, all
without modifying the custom profile. If I load a custom profile, and change one setting to experiment, don't automatically save and overwrite the custom profile. I'm just trying something and don't want to change/break a known working profile. I can understand the need to save settings to some default profile and reboot to 'run with it'. 

Give me the option to just run with the new settings or to save back to the custom profile.  

It would be useful to set one profile to factory defaults without using the reset to factory command they wipes everything. 

It would be useful to load a profile, change it, and save it as a new profile without rebooting.

nseidle, 2022-09-01:
Ok, I think I get the reasons. We've got UX problems.

Use case from serial config:

* Unit boots and loads profileX
* User makes changes via Serial
* User powers down unit

Should we save the changes? I would suggest yes, by default. If we change the default behavior to 'not save changes at power down' but require the user to enter the profile menu and press 'save me', it may catch a lot of users off guard and lead to an increase in technical support. We need to establish if most users are like you, or if they are just 'set it and forget it' type users.

If we follow other apps (not to say they are better), there are almost no save buttons. You make changes to a google doc and they are stored instantaneously. Toggle a button in Windows config and it's done.

Other use case from AP Config:

* Unit boots and loads profileX
* User makes changes via AP Config
* User presses save settings

Should we save the changes? Yes.

I'd like to get you the interface that you think is best, but we have to consider the other users.

> If I load a custom profile, and change one setting to experiment, don't automatically save and overwrite the custom profile. I'm just trying something and don't want to change/break a known working profile.

This sounds like a 'I want to copy one profile to another spot'?

> It would be useful to set one profile to factory defaults without using the reset to factory command which wipes everything.

Great idea. Can do. Moved to separate issue.

nseidle, 2022-09-01:
I missed the opening ask:

> It would be useful to be able to load, modify, and save various custom profiles without rebooting.

This is probably not possible. There are 232 settings that need to be applied. When a user switches to a new profile, we need to reconfigure the ZED, SD, Bluetooth buffers, etc. While we could attempt to do this at profile exit, it's fraught with error, and will take just as long as a reboot.

tonycanike, 2022-09-02:
Agreed there are UX issues here to work out.  And perhaps this isn't the highest priority.  And I totally get the need for a reboot when changing the operating parameters, I'm not suggesting otherwise. 

We're probably looking at "profiles" differently.   I've been thinking of a "profile" like a configuration file that can be edited to be used when desired in the future.  

I'm gathering others see "profiles" as the place the current running machine configuration is automatically saved and reloaded from.

I'm just looking for a way to setup a number of slightly different profiles without going through multiple reboot cycles.  I could use a text editor, so maybe I'll do that.

Perhaps one way of thinking about it is I would find a  "profile editor" useful, both in serial and AP.  In my mind, I'm not changing the operating parameters, I only want to setup, copy, and edit a few profile files for use in the future.  It would be useful to edit (read, modify, save)  a profile file, or copy a profile into a new file, etc., without rebooting.

But I get this might not be the direction to take things in.  

One of the needs driving this the importance of properly entering fixed base coordinates, and my desire to enter them when in a comfortable setting without blinding sunlight and biting mosquitos (office, air conditioned car) to reduce errors.   I'd like to have a handful of base coordinates ready to use for a day's work.  I'm thinking having a few profiles pre-set would be a good way to accomplish this.  I also want to verify that I properly entered the coordinates.

But there are other ways to accomplish this.   Perhaps a CSV point file on the SD card and the WiFi config could read in a point from the file for the fixed base coordinates.   I'm also going to try to get my surveyor data collector to set the fixed base coordinates (it claims to do this with the ZED).

Or I want to setup a lot of similar but different profiles for different logging and measurement rates.   In a given week, I might set my units for
1. Raw logging only for PPP static solution with reduced measurement and message rates. OPUS decimates all submissions to 30 second intervals, so why would I log every second if I'm only PPP static logging?
2. RTK Base with no logging. Default message rates out to the rover.  
3. RTK Base with logging because the previous day I got about 50 false RTK Fixes (yes Virginia, there is no Santa Clause, RTK Fixes can be off, can't always believe 0.014 even with 25 satellites, some days false fixes happen quite often, especially under tree canopy.  The big boys run multiple RTK engines with different parameters and compare the results internally,  kinda like the Space Shuttle having three computers programmed by three different programming teams and the results are compared).    I had to toss a whole day's work, fortunately I realized it.  I was dumping my receiver, going back to the same point, waiting for RTK Fix, and got different coordinates 3 times in a row on the same point, usually off by a couple feet in the northing and easting. Surveyors tell me these things happen in dense canopy.
Anyway I'm thinking of using RTK lib to post process my base & rover logs to check my fixes.
4. Rover with no logging.
5. Rover with logging. (see above)

Some of this I could do by taking out the SD card instead of different profiles, but those cards are tiny, easily lost, and sometimes they slip inside the Facet along side the SD reader instead going into the SD reader.  Then I need to find a tiny screwdriver.....

tonycanike, 2022-09-03:
I'm going to close this as it isn't directly actionable.

Having a database of points (coordinate sets) to load into the ZED for a fixed RTK base is a function covered by some commercial data collectors.  This is a common use case if a surveyor has a set of control points already established and they need to use one for the day's work.    This was one of the initial drivers for this issue report.  I'll pursue using my data collector to set the ZED fixed base coordinates (SurvPC does have some support for a generic F9P.)

Though I give a big sigh when I'm trying to setup a profile for tomorrow's work and the Facet starts trying to use it today when I'm half-way through setting up the profile.  Add that to the fast serial menu timeouts (#305) and my sighs get longer and sadder.

closed 2022-09-03 by tonycanike

gdt, 2022-09-03:
In order to be able to say "this is the data collector's job" it is necessary for the ESP32 code to have an API to set base coordinates, sort of REST with sending geojson or something, and really that should work over the bluetooth or TCP and not require messing with wifi.

Given that the ESP32 code owns the F9P, I don't think it's a good idea for other code to be writing to or configuring the F9P.