sparkfun/SparkFun_RTK_Firmware #311: build instruction buglets: required libraries

gdt, 2022-08-28:
I'm setting up to build on a mac.   I should point out that I've been writing code for a really long time, but am new to the arduino IDE.   Things I found:
  - mqtt pub sub: comment by include says it is built in but it's not (arduino for mac 1.8.19)
  - instructions do not explain where to check out the github-only libraries *to*; I figured it out
  - instructions talk about `RTK_Surveyor.ino ` but there is a library manager comment in bluetoothselect.h

I wonder if it would be possible/wise to have git submodules for all the libraries, and instructions to symlink them all into the arduino library dir, or config to use them in place.   With a submodule you'll get the intended tag, and updating is a commit to the submodule info to change the subsidiary checkout point.   It would bring more rigor (exact pinned dependencies) and make it easier for users even if there is a need to update to new versions with more thought.

nseidle, 2022-08-29:
Thanks for reading and reporting. It's minutia that make the difference.

> mqtt pub sub: comment by include says it is built in but it's not (arduino for mac 1.8.19)

Link? I [believe](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/main/Firmware/RTK_Surveyor/RTK_Surveyor.ino#L154) we updated the comment

> instructions do not explain where to check out the github-only libraries to; I figured it out

Being someone who swims in Arduino daily, I guess I don't understand the confusion. I welcome a PR on the docs or a recommendation. For non-Arduino users would a link to [How to install an Arduino Library](https://learn.sparkfun.com/tutorials/installing-an-arduino-library/all) help?

> instructions talk about RTK_Surveyor.ino but there is a library manager comment in bluetoothselect.h

A link would help clarify. [These instructions](https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update/#windows_1) list the various libraries and where to get them. I suspect you are talking about the [Ubuntu](https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update/#ubuntu-2004) instructions?

> I wonder if it would be possible/wise to have git submodules for all the libraries, and instructions to symlink them all

I'd rather not go down that route. [I had a bad experience](https://www.youtube.com/watch?v=A4_OJ2bciRc&ab_channel=BobVincent) with submodules, but I can be convinced. Please note (thank *you* for the prodding to create it, btw) that the [yaml build](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/main/.github/workflows/compile-release.yml) is the voice of truth WRT the various libraries and version numbers of those libraries for firmware compilation. Currently we are only setting the ESP32 core version; we are not working off any specific library version.

gdt, 2022-08-29:
You are correct that the mqtt "builtin" is fixed; I was on release_candidate a bit ago.  Sorry for the noise.

Yes, a pointer to that tutorial would help newbies a lot and not take up a lot of space.   I am in the position of starting to use the arduino IDE because of this code, and I don't know how odd that is.

Perhaps point to the yaml build in the how-to compile as some data that may be useful?

I don't mean to push hard for submodules.   My bias  is really to more scripted and less clickly, and to pinned dependencies at least as an option.   But maybe this is not a big problem in practice and time is better spent elsewehere.

gdt, 2022-08-29:
The yaml build file is a great hint that one can run the arduino ide as a cli with no display, and that's great.  That's my bias anyway.   Also could be in the docs, could be as simple as:

One can use the Arduino CLI, and instructions for doing that on GNU/Linux, including fetching libraries, are implicitly given in the CI build recipe in [pointer].

nseidle, 2022-08-31:
Are you following the [windows](https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update/#windows_1) or the [ubuntu instructions](https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update/#ubuntu-2004) as your primary source for Mac install? I ask because the Windows instructions already include a "Note: You should click on the link next to each of the #includes at the top of RTK_Surveyor.ino within the Arduino IDE to open the library manager and download them."

We really should write a separate instructions for Mac. If you feel like sketching something and dropping it here, I'll get it into the docs. If you want to go for bonus points and doing a PR, I'd welcome that as well.

gdt, 2022-09-18:
I was following windows, because I didn't realize there were ubuntu, but now that I do, I wlll look at that.  I did see the "click link next to includes", but found some discrepancies which I think you have fixed.

My confusion was that I had zero idea were the arduino world put libaries, but with a link to a library tutorial that should be ok.  I get it that it's not special about this.  (I find a central location odd as surely sometimes different projects need different versions.)

I'll see about improvements, but it doesn't seem useful to keep this open.  Thanks for the fixes and comments.

closed 2022-09-18 by gdt