sparkfun/SparkFun_RTK_Firmware #316: Firmware update only from CLI

pyrog, 2022-08-31:
### Subject of the issue
Can't update the firmware from web interface or USB.
This is a known issue (can't find it) . I understood that it will be closed since 2.4 RC.

### Your workbench
* RTK Surveyor 4Mb
* Actual firmware: RTK_Surveyor_Firmware_RC-Aug_11_2022.bin
* New firmware: [RTK_Surveyor_Firmware_v2_4.bin](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/download/v2.4/RTK_Surveyor_Firmware_v2_4.bin)

### Steps to reproduce
Try to update from USB or web but the Surveyor reboot instantly.
The device crash and display on USB:
```
Menu: Update Firmware Menu
1) Load RTK_Surveyor_Firmware_v2_4.bin

2) Load RTK_Surveyor_Firmware_v2_4.bin

x) Exit
1
Loading RTK_Surveyor_Firmware_v2_4.bin

Moving file to OTA section
Bytes to write: 2052352
[=0%
abort() was called at PC 0x400826e6 on core 1

Backtrace:0x40083dad:0x3ffd13300x40095369:0x3ffd1350 0x4009a8d1:0x3ffd1370 0x400826e6:0x3ffd13f0 0x4011f95f:0x3ffd1440 0x40109e19:0x3ffd1460 0x40101392:0x3ffd1480 0x40101539:0x3ffd14a0 0x400e037e:0x3ffd14d0 0x400e05eb:0x3ffd1d70 0x400e7ed7:0x3ffd1d90 0x400e7f9e:0x3ffd1db0 0x400e7fd3:0x3ffd1dd0 0x4010d9c1:0x3ffd1df0 

ELF file SHA256: 0000000000000000

Rebooting…
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
Display not detected
ZED-F9P firmware: HPG 1.13

LittleFS Started
SparkFun RTK Surveyor v2.4-Aug 27 2022
Unknown setting espnowPeers.5
Unknown setting espnowPeers.6
Unknown setting espnowPeers.7
…
Unknown setting espnowPeers.66
Unknown setting espnowPeers.67
Reset reason: ESP_RST_PANIC
Unknown: ._RTK_Surveyor_Firmware_v2_4.bin

microSD: Online
microSD: Offline
Unknown: ._RTK_Surveyor_Firmware_v2_4.bin

microSD: Online
Unknown setting espnowPeers.5
Unknown setting espnowPeers.6
…
Unknown setting espnowPeers.67
Warning: Settings size is 4488 but current firmware expects 4496. Attempting to use settings from file.
Unknown setting throttleDuringSPPCongestion
Unknown setting enableNtripServerMessageParsing
Unknown setting espnowPeers.5
Unknown setting espnowPeers.6
…
Unknown setting espnowPeers.67
Profile ‘XYZ’ loaded
```

nseidle, 2022-08-31:
Thanks for reporting! 

This is a known issue when updating a unit from v1.x firmware to v2.x. You're going from RC-Aug-11 but it's not clear if RC-Aug-11 was loaded with SD or other. Either way, I'll try to duplicate in the coming days. For now, use the GUI or CLI and let me know if you still can't update. My go to is the GUI these days. It's just so quick and easy.

pyrog, 2022-08-31:
RC-Aug-11 was **loaded from CLI** under macOS.

Can't use GUI (windows app) from macOS 😉

nseidle, 2022-08-31:
> Can't use GUI (windows app) from macOS 😉

But you can use the python GUI :)

> RC-Aug-11 was loaded from CLI under macOS.

Got it. And just to be sure, you pointed at the [16MB partition bin](https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update/#macos-linux)?

    py esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 ./bin/RTK_Surveyor.ino.bootloader.bin 0x8000 ./bin/RTK_Surveyor_Partitions_16MB.bin 0xe000 ./bin/boot_app0.bin 0x10000 ./RTK_Surveyor_Firmware_vxx.bin

nseidle, 2022-09-01:
Dah! Sorry, I failed to read you have a 4MB surveyor.

Yes. I can confirm your issue. This is because there is not enough flash space to load new firmware over OTA. Basically, 4MB Surveyors must have new firmware loaded over GUI or CLI. SD and WiFi are not supported. This is documented [here](https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update/#updating-4mb-surveyors). 

But we should fail gracefully! I will add a message to the user if firmware upgrade via SD on 4MB units is not possible.

Thank you for reporting.

nseidle, 2022-09-01:
Ok, fix is in place. User is notified with a message 

> SD firmware updates are not available on 4MB devices. Please use the GUI or CLI update methods.

if the device is not capable of firmware update via SD.

pyrog, 2022-09-02:
> Basically, 4MB Surveyors must have new firmware loaded over GUI or CLI.
Ok, thanks.

Previous test on 2.4RC fails too with RTK Express. I will check in few days.